node-opcua/node-opcua #132: Manipulate Status Codes

immenz, 2015-10-27:
Hello, 

where can i manipulate the status codes of the simulation nodes in the Server? I try to get random codes between good and bad.

erossignon, 2015-12-02:
StatusCodes are currently expressed as enum 
see node-opcua\lib\datamodel\opcua_status_code.js, node-opcua\lib\raw_status_codes.js
This is convenient but doesn't allow any statuscode to be easily written.

What sort of missing StatusCode do you need ? Can you elaborate.

erossignon, 2016-09-07:
StatusCode can be enriched with Status bits now ( see makeStatusCode )
https://github.com/node-opcua/node-opcua/blob/master/lib/datamodel/opcua_status_code.js#L12
https://github.com/node-opcua/node-opcua/blob/master/lib/datamodel/opcua_status_code.js#L307

and example of use:
https://github.com/node-opcua/node-opcua/blob/f59c35b774b944aca370f9ab02eb47b27327daff/test/datamodel/test_status_code.js#L105

closed 2016-09-07 by erossignon