node-opcua/node-opcua #134: create a server & client

devilFruit007, 2015-10-30:
Hello everbody,

I have a problem. I want to send a simple message from server to client with the document JSON. The Client must receive the document without problem und must show the message.How does that function?

erossignon, 2015-11-01:
Can you clarify what you mean by 'document JSON' ? do you want to exchange a string between client & server ?

devilFruit007, 2015-11-01:
yes, exactly!! I want to exchange a string message between client and server.

erossignon, 2015-11-01:
Are you developing a Server or a Client ?

devilFruit007, 2015-11-01:
I’m trying to develop a Server AND a Client. The Server must build a connection and must send a string message to the Client. The Client must receive the string message und must show at the command prompt.
For example, I have 2 command prompts. The first prompt is for the server. I'm typing in the command prompt “node sample_server.js. The second prompt is for the client. I'm activating the client with “node sample_client.js”. The client must receive the message such as “Hello World”.

erossignon, 2015-11-05:
you could expose a String variable in the server and monitor it in the client.

```
        server.jsonVar= server.engine.addVariable("MyDevice", {
            browseName: "JSONObject",
            dataType: "String",
            value: {
                get: function () {return new opcua.Variant({dataType: opcua.DataType.String, value: get_json_string() });}
            }
        });
```

devilFruit007, 2015-12-28:
Thank you for your response in your useful tips. Unfortunately it is not function. May you please help me how it monitor in the client?

closed 2016-03-19 by erossignon

RakshanPremsagar, 2019-05-06:
just with the sample_server and sample_client javascript files, how to send a message from client to server?
Please, a simple small code would help me understand
PS: Total newbie to OPCUA