node-opcua/node-opcua #135: Add the ability to limit the number of monitoredItem for small embedded server

erossignon, 2015-11-01:
We need to add the ability to restrict the number of monitored items a client can create to protect nano server from being overloaded.

erleeman, 2016-02-17:
Hi @erossignon 

What would be the right place to implement this?
I am new with node / node-opcua but would like to support.

konradheidrich, 2016-02-17:
Maybe I am wrong, but monitored items needs subscription and subscription is not part of Nano profile - it is implemented in Micro? AFAIK this is the main difference between Nano and Micro and related to many services which needs to be implemented to the server.

erleeman, 2016-02-22:
I am currently working on this but the currentMonitoredItemsCount per Session is always 0. Is this a already known bug?

erossignon, 2016-03-05:
@erleeman  currentMonitoredItemsCount is currenly not wired and always returns 0. This is work in progress.

closed 2018-01-25 by erossignon