node-opcua/node-opcua #137: Reading and Monitoring opaque Variables

drealie, 2015-11-17:
Hi, 

i have a OPC UA Server which has the Variable shown in UAExpert like:

NamespaceIndex : 2
IndentifierType :Opaque
Identifier: len=5, 0xa013000013

the readFunctions "readVariableValue(ns:2;b=0xa013000013)..." of my Client is returning:

StatusCode {
       value: 2150891520,
       description: 'The node id refers to a node that does not exist in the server address space.',
       name: 'BadNodeIdUnknown' },

The browse Result of the Numeric Node of the containg folder shows: 

NodeId {
  identifierType: { [Number: 4] key: 'BYTESTRING', value: 4 },
  value: <Buffer a0 13 00 00 13>,
  namespace: 2,
  namespaceUri: null,
  serverIndex: 0 }

Any Ideas how i can properly get read and monitoring for the this Variable?

erossignon, 2015-11-17:
try ``` readVariableValue("ns:2;b=a013000013" ...``` (without ```0x```)
(see   https://github.com/node-opcua/node-opcua/blob/master/test/datamodel/test_nodeid.js#L93 for an example)

drealie, 2015-11-18:
thank you for your prompt answer. 

via the first method i get 

StatusCode {
       value: 2150891520,
       description: 'The node id refers to a node that does not exist in the server address space.',
       name: 'BadNodeIdUnknown' },

interestingly https://github.com/node-opcua/node-opcua/blob/master/lib/datamodel/nodeid.js#L261 returns a <b>value of []</b> for <b>0xa013000013</b> and a <b>value of [[0=160],[1=19],[2,=0],[3=0],[4=19]]</b> for <b>a013000013</b>

 via the second method for both <b>0xa013000013</b> and <b>a013000013</b> i get: 

[TypeError: value is out of bounds]
TypeError: value is out of bounds
    at checkInt (buffer.js:792:11)
    at Buffer.writeUInt32BE (buffer.js:881:5)

erossignon, 2015-11-25:
@drealie 
I would be happy to help but can you modify your previous post to be more specific and provide the exact code that you used for 'first method' and  'second method'.

closed 2016-07-07 by erossignon