node-opcua/node-opcua #138: Detecting when a session has expired.

Guardiannw, 2015-11-19:
Good Morning Etienne,
    I am writing a REST application to interface with the Kepserver OPC UA Server.  For speed sake, I am trying to keep a single active connection and session between all of the different REST requests.  I am having a problem with the session timing out on me without any way for me to monitor it and try to reconnect when it fails.  Right now the only way I can get this to work is to try another read request with the UA server and try to recreate a session if the error code is 2149908480.  Is there any way that I can subscribe to some sort of event that will notify my application that the session has timed out?

Thanks in advance,

Guardiannw, 2015-11-19:
F.Y.I., I am choosing not to reestablish a connection and session for each request for 2 different reasons.  
    1. There is a significant latency overhead cost.
    2. I get a buffer out of bounds error when I get many requests back to back/simultaneous.

Thanks again,

erossignon, 2015-11-19:
One way is to setup a timer that activate the connection with the server by reading the ServerStatus for instance.
I have the same issue with the interactive_client, when the user may be pausing for a while before entering the next command, causing the connection to time out
see 
https://github.com/node-opcua/node-opcua/blob/master/bin/interactive_client.js#L281

I hope this will help.

Guardiannw, 2015-11-20:
Thanks for the quick reply, I am working on an implementation of your proposal right now.  Will respond with my results.  Thanks again,

Guardiannw, 2015-11-23:
Thanks again @erossignon, the method you suggested worked like a charm!

closed 2015-11-25 by erossignon