node-opcua/node-opcua #140: Establishing a secure connection

Nestoteles, 2015-11-25:
Hey,
i started to play around with different security modes and policies, but didn't manage to establish a connection. I used the sample client and server and tried to create certificates as described in the 'create a server' tutorial.
Is there an example which uses MessageSecurityMode, SecurityPolicy != none or can you explain how to create and pass the certificates to client and server?

Thanks in advance, 
Nestoteles

Nestoteles, 2015-11-30:
after some more researsch this seems to be related to #127

Nestoteles, 2015-12-07:
I'm still having problems with this after realese of version 0.0.50.
I try to establish a connection with SecurityMode: SignAndEncrypt and securityPolicy: Basic128Rsa15
What i did

- created sample server and client which work with no security mode and policy mode: None
- created selfsigned certificates as described in notes_on_certificates.md
($ openssl req -x509 -days 365 -nodes -newkey rsa:1024 -keyout key.pem -out cert.pem)
- copied the certificate of the server to the client in a new folder "trusted_certificates"
- changed options of client and server to run with certificates, SecurityMode: SignAndEncrypt and securityPolicy: Basic128Rsa15

Client Code:
```
var options = {
    securityMode: MessageSecurityMode.SIGNANDENCRYPT,
    securityPolicy: SecurityPolicy.Basic128Rsa15,
    requestedSessionTimeout: 10000,
    applicationName: "NodeOPCUA-Client",
    endpoint_must_exist: false,
    certificateFile: "certificates/cert.pem",
    privateKeyFile: "certificates/key.pem",
    serverCertificate: crypto_utils.readCertificate("trusted_certificates/cert.pem")
};

var client = new opcua.OPCUAClient(options);
```

Server Code:
```
var server = new opcua.OPCUAServer({
    port: 4334, // the port of the listening socket of the server
    resourcePath: "UA/MyLittleServer", // this path will be added to the endpoint resource name
    certificateFile: "certificates/cert.pem",
    privateKeyFile: "certificates/key.pem",
    buildInfo : {
        productName: "MySampleServer1",
        buildNumber: "7658",
        buildDate: new Date(2014,5,2)
    }
});
```

When the client tries to create a session, i recieve the following error:

SampleClient\node_modules\node-opcua\lib\client\opcua_client.js:100
        applicationUri = e.tbsCertificate.extensions.subjectAltName.uniformResourceIdentifier[0];

TypeError: Cannot read property 'uniformResourceIdentifier' of undefined

Did I make a mistake when creating the certificates or is the way i pass them as arguments not correct?

Tanks in advance,
Nestoteles

erossignon, 2015-12-07:
@Nestoteles the tutorial is outdated and need to be updated.
(openssl req -x509 is not enough as we need a V3 x509. This requires a open ssl config files=)

When installed, node-opcua create a bunch of sample certificates ( primarily used for testing) 
you can find them in the certificate folder and reuse them.

see bin\simpe_server.js and bin\simple_client for an example.

Nestoteles, 2015-12-10:
Thanks a lot. One more question, is there a possibility to pass the serverCertificate to the client from the file system?
Something like i tried in my example with:
```
var options = {
    ...
    serverCertificate: crypto_utils.readCertificate("trusted_certificates/cert.pem")
};

var client = new opcua.OPCUAClient(options);
```

This leads to an error, even with the correct certificates.
Client output:
 cannot connect to endpoint : opc.tcp://PC006:4334/UA/MyLittleServer
 failure  [Error: BadCertificateInvalid (0x80120000)]
Server output:
 Invalid receiver certificate thumbprint : the thumbprint doesn't match server certificate !

erossignon, 2015-12-12:
It should be possible to do it. can you confirm that the cert.pem file is an ascii file looking like:
```
-----BEGIN CERTIFICATE-----
...base 64 string
-----END CERTIFICATE-----
```

Nestoteles, 2015-12-14:
I tried the server_cert_1024.pem from the sample certificates and used it as serverCertificate in client and as certificateFile in the server. It has more information than just the base 64 string.
I tried deleting everything except one of the base 64 strings. Doing so leads to an error: 
"failure  malformed X.509 certificate PEM (code:001)"

Something i noticed is that the certificates that are recieved from the server and saved with
```
var certitifacte = endpoint.serverCertificate;
var certificate_filename = "server_certificate" + i + ".pem";
fs.writeFile(certificate_filename, crypto_utils.toPem(certificate, "CERTIFICATE"));
```
only have the 
-----BEGIN CERTIFICATE-----
...base 64 string
-----END CERTIFICATE-----
part. Even though the server uses the same server_cert_1024.pem as mentioned above.

erossignon, 2015-12-22:
certificates in the node-opcua folder should be ready to be used without modification by node-opcua client/server. The PEM format can contains extra information that explained what is in the Base64 string.

Nestoteles, 2015-12-22:
Thanks for your effort. Unfortunately i'm still struggling with this.
Connecting to a server and and getting the serverCertificate from the endpoints is working fine. But to do this the server needs to allow unsecured communication, which I dont want.
I want that only Clients that have the serverCertificate already before connecting are able to establish a connection to the server. So that an admin is in control of certificate distribution.
Is using ``` crypto_utils.readCertificate("trusted_certificates/copyOfServerCertificate.pem"  ``` the correct way of loading the certificate?

Nestoteles, 2015-12-29:
@erossignon: One question, i see that the part with certificates is still work in progress. 
I was looking for something like the trusted list mentioned in the Security specification 4.10 Application Authority to manage Certificates. Is there already a possiblity to do this with node-opcua?

erossignon, 2016-01-18:
Please bare with us, we're working on it.

closed 2016-12-28 by erossignon

stevewillard, 2018-02-07:
@Nestoteles did you ever get this to work? Having similar issues with
```
crypto_utils.readCertificate("trusted_certificates/copyOfServerCertificate.pem")
```