node-opcua/node-opcua #142: Server create variables in loop

codanator, 2015-12-02:
Hey,

i want to create a server with variables created from an config.file.

In my loop i have to set a variable for every node that getter and setter can use.
But how can i set a variable that getter and setter can use in their callbacks.

I have tried to save a variable with nodeid as the key.
In the getter this is ok because i have this.nodeId.value but in setter this is the nodes global object.

Here is an examle:

```javascript
function createAdressSpace(server) {

    var rbgs = spsvars.rbgs;
    var nodes = spsvars.nodes;

    for (var key in rbgs) {
      var value = rbgs[key];
      var rootNode = value.rootNode.replace('ns=4;s=', '');
      server.engine.addFolder("RootFolder", {browseName: rootNode});

      for (var node in nodes) {
        var n = nodes[node];

        var variable = n.value;

        var serverNode = {
           nodeId: "ns=4;s=" + rootNode + '.' + n.nodeId,
           browseName: node,
           dataType: "Boolean",
           value: {
             get: function () {
               //here the variable is always the last that was set in loop - naturally
               return new opcua.Variant({dataType: opcua.DataType.Boolean, value: variable});
             },

             set: function(variant) {
               //here the variable is always the last that was set in loop - naturally
               variable = variant.value;
               return opcua.StatusCodes.Good;
             }
         }
        }

        server.vars = server.engine.addVariable(rootNode, serverNode);
      }
    }

}
```

What is the best way to get this working?

Thanks in advance, i really like this project and the great work you have done

erossignon, 2015-12-02:
Mario, you can solve your problem by taking advantage of the closure concept of javascript.
Have a look at http://javascriptissexy.com/understand-javascript-closures-with-ease/ and try to sort out the issue by yourself. Let me know if you're still stuck and will give you a hint.

codanator, 2015-12-02:
Ohh... that was easy!

for (var node in nodes) {
        var n = nodes[node];
        var variable = n.value;

        var serverNode = function(variable) {
          var serverNode = {
             nodeId: "ns=4;s=" + rootNode + '.' + n.nodeId,
             browseName: node,
             dataType: "Boolean",
             value: {
               get: function () {
                 return new opcua.Variant({dataType: opcua.DataType.Boolean, value: variable});
               },

               set: function(variant) {
                 variable = variant.value;
                 return opcua.StatusCodes.Good;
               }
           }
         }
         return serverNode;
        }(variable);

        server.vars = server.engine.addVariable(rootNode, serverNode);
      }

Thank you very much!!

closed 2015-12-02 by codanator