node-opcua/node-opcua #143: Best Practice: Read/Write in Client

Nestoteles, 2015-12-14:
Hey, I am working on a control circuit with node-opcua.
I have client with a few sensors and actuators. The sensor data have to be sent to the server every second and other values, have to be read to control the actuators.
My question is, what is the suggested best practice for a client that is writing and reading/subscribing?
I am new to node.js and would be happy for all kind of tips and tricks.

erossignon, 2015-12-22:
@Nestoteles : it looks like there is a client/server mismatch in your description:

In typical use, we would implement a server on the sensors and actuator side.
The server exposes the sensor values as variable nodes.
The client  subscribes to the sensors value either using reading or subscribing.
When creating subscription and monitored item, the client negotiates the sampling interval with the server ( every seconds)
The server can also provides "methods" so the client can activate the actuators remotely.

Nestoteles, 2015-12-22:
Thanks, I already found a solution, but swapping server and client, will make things even easier.

closed 2015-12-22 by Nestoteles