node-opcua/node-opcua #144: Client reconnection concept after unexpected socket close (connection break)

pelis, 2015-12-15:
I am trying to figure out how to best reconnect after the socket connection in OPCClient closes in case of a physical connection error.
What I am currently doing:
-	Close the current Client session (clientSession.close())
-	Disconnect - opcClient.disconnect()
-	create repeated timer trying to reconnect which succeeds when the physical connection to the server is up again.
Problems I am facing:
opcSession.close() and opcClient.disconnect() have rather long lasting timeouts when the physical connection is broken. In this case timeoutHint is set to 0 – would be great if this timeout could be set or would be rather short so that reconnect could be triggered quicker.
Are there plans to implement a reconnect concept? Maybe I am just not finding it. OPCClient seems not to support a reconnect() or even an auto-reconnect with an optional interval. That would be a great enhancement I think…
For now my above-described implementation works OK – the only drawback is the long disconnect timeout – which potentially leads to data loss (I am still not willing to create new OPCClient to get around it and I have not yet tried it.)
One other small suggestion:
It would be nice to be able to pass a session name – when you want to do some server monitoring…

pelis, 2015-12-16:
One more question regarding sequence number / data integrity
How would I best recover the data being queued on the server while the connection is down?
Is there a way to 're-subscribe' based on a sequence number?
Or would I have try to read historic data first (based on the timestamps from the last items timestamp before connection loss)

erossignon, 2015-12-22:
OPCUA provides a mechanism to allow client to reconnect to a server without loosing the data. 
When a client disconnect unexpectedly, the server maintains the client session alive until it times out.
This allow the client to reestablish the connection and reconnect to the same session again.
OPCUAClient#reactivateSession() can be used to transfer the old session to the new client connection. (see https://github.com/node-opcua/node-opcua/blob/master/test/end_to_end/u_test_e2e_transfer_session.js#L264 for a example ).

While the client is disconnected and the session is still alive, the server still processes the data change notification on monitored items. Unprocessed subscription notification will be sent to the new client once the connection is established.

erossignon, 2015-12-22:
How the client should cope in case of a physical connection lost: Your use case is interesting and shall be turned into a automated test so we could investigate and fine tune the behavior in this situation.

pelis, 2016-01-05:
Well, the client should try to reconnect. Ideally, - as you have suggested above - reactivate the Session. Should that fail, create a new session and re-subscribe (even though data would probably get lost).
I have not yet looked into the timestamps, whether or not that could help to read historic data beginning at the last time-stamp when the client has received data for the last time. This is an option when the session cannot be reactivated. If that would work, the subscription could provide a function for that purpose. Such function would have to consider, while historic data is processed, new incoming data would have to be queued.

Please see also my comment in #141 >>

erossignon, 2016-01-07:
OPCUA specifies how reconnection should happen:
```
Release 1.03 106 OPC Unified Architecture, Part 4 
6.5 Re-establishing connections 
```

pelis, 2016-01-08:
OK - I have to admit, that I did not read the spec here...
However - I think this is not too far away...
Looking at the (low-level) socket seems to me still appropriate.
Republish ist still completely missing.

erossignon, 2016-01-18:
@pelis , I have added re-connection support to the client.  let me know if this new implementation is OK.
You should not have to worry anymore about re-establishing the session or repairing the subscription your self. In case of connection break shorted that the subscription lifetime, you should not loose any monitored item values.

pelis, 2016-01-19:
Thanks! Quiet a lot of code!
I will need some time to play with it - I will certainly let you know once I am done
Cheers

P.S Please feel free to close #141

pelis, 2016-01-28:
I have now done quite a bit of testing.
Everything works fine when the connection gets interrupted for a certain period of time – great job!
-	Session and subscriptions are maintained, as expected
-	Transport layer / socket is a new one.

So when I subscribe to “socket.on(“close”) I will get called in the “close” handler. I can get the new active socket again (via opcclient._secureChannel._transport._socket).
As you have noted in your code, when the server gets restarted, you are forcibly closing all sessions – it would help when you could fire a session “closed” event and/or subscription “destroyed” or “inactive”, or instead, see below comments.
It seems that the “subscription.isActive()” issue has returned – it always returns true (even when the session has been closed forcibly).
It is great that you have solved the “resubscribe” issue and it is fine for me to take care about creating a new session and new subscriptions when the session is closed. However, I have to implement a way how I can be absolutely sure that the “republish” loop has ended so I would not get in conflict with it. A good way to check that could be when subscription.isActive() would return false when the session has closed and/or when the session would also provide a method like “session.getStatus()”. Session status codes could be “active”, “interrupted”, “closed”, where “interrupted” means that you are trying republish the session.
In a server restart scenario, it would also be important to know when the “create new session, create new subscriptions” would expire, so that the client programmer (in this case me) does not run in a conflict when trying to reconnect.

pelis, 2016-03-07:
Disconnecting during reconnect (backoff)
Currently it seems that the backoff process trying to reconnect will continue even when OPCUAClient.disconnect() is called. I think OPCUAClient.disconnect() should kill the reconnect process. Now it waits connectionStrategy.maxRetry until it finally gives up. This also sometimes leads to some ugly null pointer exceptions (after .disconnect(), not reproducible ).
Especially when connecting via OPCUAClient.connect(endpoint, function (err) {}) and the server is not available, the callback - function(err) - is called after “reconnect()” gives up. Per default setting this is 100 * 20000 (maxRetry *  maxDelay) which computes to ~ 33 Minutes.
Is there a recommended way to stop the reconnect process or to kill the connection (so that the connect callback is called right away and "everything" is clean there after?

One note on the documentation:
You may want to add the "backoff” event to the OPCUAClient events…

erossignon, 2016-03-08:
@pelis , I am proposing to add the following test case scenari ( Given/When/Then ) to trap the issues above.
```
    it("TR8 -  disconnecting during connect", function() {
        // Given a client that has a infinite connection retry strategy,
        //   And that client#connect is call to connect to an non-existent server.
        //
        //  When client#disconnect is called
        //
        //  Then the client should complete the client#connect async call with and err;
        //   And the client should stop the automatic reconnection strategy (backoff)

    });
    it("TR9 -  disconnecting during reconnect", function() {
        // Given a client that has a infinite connection retry strategy,
        //   And the client has a lived connection with a server
        //   And that the connection has dropped ( backoff stragety taking place)
        //
        //  When client#disconnect is called
        //
        //   And the client should stop the  automatic reconnection strategy (backoff)

    });
    it("TR10 -  a client should notify that the reconnection attempt is taking place with an event", function() {
        // Given a client and a server with an established connection
        // When the connection link dropped
        // Then the client shall raise an event to indicate that the reconnection process is now taking place.
    });
```
Let me know if I miss something

pelis, 2016-03-16:
Thanks for the new code! – Almost there, here is my testing results:
Case TR8: OPC-Server NOT running or reachable.
When client.disconnect is called "_cancelBackoff” is set
> Works as expected

Case: TR9.1OPC-Server running, connection established, then OPC-Server is stopped
When client.disconnect is called "_cancelBackoff” is set
> Works as expected, however in client_secure_channel_layer.js timer is still active:
timerId = setTimeout(function () {
        console.log(" Timeout .... waiting for response for ", requestMessage.requestHeader.toString());

Case: TR9.2 OPC-Server running, connection established, then physical connection is interrupted
> Still not working for me, “_cancelBackoff” not triggered, backoff process keeps trying…
What I am currently doing as a workaround is - just before I call client.disconnect():
client._secureChannel.__call._cancelBackoff = true;

erossignon, 2016-03-19:
@pelis, I have got trouble to reproduce. Would you be able to create a new unit test to highlight the issue ?
Unit tests can be run this way, btw
```
$ npm install mocha -g
$ cd node-opcua
$ mocha test\end_to_end\test_e2e_connection_reconnection.js
```

pelis, 2016-03-21:
I did run the unit tests which work fine - however, in real live the above issues arise.
I will come up with something after Easter holiday.

erossignon, 2016-04-16:
@pelis, can we consider this closed as it as been released in 0.0.53? ( we still can reopen a new issue if we find a another problematic use case)

pelis, 2016-04-18:
yes, let's close it.
I have thought about coming up with unit test but did not yet dive in simulating socket problems...
I think there still some issues in the timer handling, but let's just close it now

closed 2016-04-18 by pelis