node-opcua/node-opcua #149: publish-subscribe

devilFruit007, 2015-12-29:
Hello,

I have a question: Does node-opcua support the function publish-subscribe model? If yes, how does it function?

erossignon, 2016-01-07:
Subscription in OPCUA are quite different from the PUB/SUB concept in MQTT.
You can find a useful document on how subscription works in OPCUA 
http://documentation.unified-automation.com/uasdkc/1.4.0/html/L2UaSubscription.html

ozzzzchen, 2016-07-15:
The new and enhanced for OPC UA Publish-Subscribe (Pub/Sub) will suit to IIoT.
[https://www.unified-automation.com/news/news-details/article/1217-opc-ua-is-enhanced-for-publish-subscribe-pubsub.html]

erossignon, 2019-03-03:
duplicates #571

closed 2019-03-03 by erossignon

erossignon, 2022-08-13:
NodeOPCUA PubSub Module is available as a commercial module from Sterfive.
An evaluation version has been published as a npm modules

https://www.npmjs.com/package/node-opcua-pubsub-server
https://www.npmjs.com/package/node-opcua-pubsub-client

npm install node-opcua-pubsub-server
See https://node-opcua.github.io/tutorial/2022/02/16/node-opcua-pubsub-episode1.html for more info

Full version, including source -code and npm modules are available for customers through [Sterfive](mailto:contact@sterfive.com) 's private repository