node-opcua/node-opcua #151: [client] FIN, ACK is missing after 'CloseSecureChannel' message

zombinary, 2016-01-04:
I will close the Secure channel to reconnect them but this takes a long time.
After an 'CloseSecureChannel message' i get an 'Timeout .... waiting for response for...' in the console.

In comparison with the UaExpert, the Client transmit an 'FIN, ACK' and runs very fast.

I couldn't find a way to send a 'FIN, ACK'.
How i can send a FIN, ACK by using the tcp-transport layer?

node-js client connect to OpcUa-Server:
[image]

UaExpert connect to OpcUa-Server:
[image]

erossignon, 2016-01-06:
I can you provide a gist of you connection reconnection sequence ?

zombinary, 2016-01-07:
To handle several endpoints with one client, i build an event-handler.
This async-process is running by firing an 'connect event'. 

```
OPCUAClient.prototype.connect = function(userIdentity) {
			
	if(!this.connected) {
       var self = this;
       
		async.series([
				/* connect endpoint */
				function (callback){
						if(self.connectionError){
							/* create new client if BadSession occured */
		       			self.client = new opcua.OPCUAClient(this.options);
		       			util.log('[erro]'+' [opcua]'+' lost connection:\ '.cyan.bold+ 'open new client');
		       		}
						self.client.connect(self.endpoint, callback);
						util.log('[info]'+' [opcua]'+' connect:\ '.cyan.bold+ self.endpoint.toString().grey);
				},
				/* disconnect client */
				function (callback){
					self.client.disconnect(callback);
					util.log('[info]'+' [opcua]'+' disconnect:\ '.cyan.bold+ self.endpoint.toString().grey);
				},       
				/* reconnect to endpoint */		       
				function (callback){
						self.client = new opcua.OPCUAClient(this.option);
						self.client.connect(self.endpoint, callback);
						util.log('[info]'+' [opcua]'+' new connection:\ '.cyan.bold+ self.endpoint.toString().grey);
		
				}
		], function (err) {
				if(err){
					util.log('[erro]'+' [opcua]'+' connect:\ '.cyan.bold+ inspect(err));
					//console.log('err:\ ', err);
					if(err != '[Error: connect already called]'){
						self.connected = false;
		           	clearInterval(self.keepAlive);
						//console.log('connectionlost'.red.bold);
						self.emit('connectionlost',err);
					}else{
						clearInterval(self.reconInt);
						self.connected = true;
					}
					return;				
			    }else{
			    	util.log('[info]'+' [opcua]'+' connect:\ '.cyan.bold+ self.endpoint.toString().blue);
			    	self.connected = true;
			    	self.reconnect = false;
					self.connectionError = false;
			   	clearInterval(self.reconInt);
					self.emit('connect');				//set node-red node status 'connected'
					self.emit('createSession');
					if(self.isConnected()){
						self.emit('keepAlive');
		}
	}
});
```

erossignon, 2016-01-07:
Interesting, I thought originally that your CloseSecureChannel OpenSecureChannel scenario was related to #144 (reopenning a secure channel after a connection break).
I have a few more question for you:
  - why would you need to handle several endpoints with one client ? can you explain the scenario here ? why not multiple client instance instead ?
 -  in your code : where is self.connectionError set to true ?

zombinary, 2016-01-08:
* why would you need to handle several endpoints with one client ? 
can you explain the scenario here ? 
why not multiple client instance instead ?
  
 * As an example for the scenario is a production hall with several machines. Every machine has his own server to 
read the sensor data.
So it is a comfortable solution to manage several servers and monitoring data with one client.
         
* in your code : where is self.connectionError set to true ?

 * This value is set true after a session connection.

* OPCUAClient.prototype.connect = function(userIdentity) 

 * This is a function with the same name in my source.
It was founded by creating an eventhandler to manage several clients.
I was unsure with this, but i found no better solution than these.

erossignon, 2016-01-09:
I found the issue that causes the long delay for the secureChannel to react on a connection break. This shall be fixed with #144.

closed 2016-03-19 by erossignon