node-opcua/node-opcua #154: Exception in server when client tries to connect

gunnarleffler, 2016-01-06:
Hi,

Firstly: love this project! I've been using v0.48 with much success. I tried to upgrade my project to v0.51 and 0.50, and found a show stopping bug.  When Inductive Automation Ignition v7.7 tries to connect to the server I get an Exception. I did notice the server API change and updated my project accordingly.

Below is the output when using `node.exe bin/simple_server` windows 7, node v4.2.4 and `set  DEBUG=ALL`.

Thanks in advance for any assistance or bugfix.

```
 Sequence Number =  2
-------------------------------------------------------- 2 GetEndpointsRequest
13:55:06:677 0 GetEndpointsRequest  ID = 2
EXCEPTION CAUGHT WHILE PROCESSING REQUEST !! GetEndpointsRequest
{ /*GetEndpointsRequest*/
 requestHeader                 /* RequestHeader                    */: {
   authenticationToken         /* NodeId                           */: ns=0;i=0
   timeStamp                   /* UtcTime                          */: 2016-01-0
6T21:55:06.677Z
   requestHandle               /* IntegerId                        */: 0
       0x0
   returnDiagnostics           /* UInt32                           */: 0
       0x0
   auditEntryId                /* UAString                         */:
   timeoutHint                 /* UInt32                           */: 5000
          0x1388
   additionalHeader            /* ExtensionObject                  */: null
 }
 endpointUrl                   /* String                           */: opc.tcp:/
/127.0.0.1:26543
 localeIds                     /* LocaleId                      [] */: null[]
 profileUris                   /* String                        [] */: null[]
};
 display_trace_from_this_projet_only =
Cannot read property 'length' of null

 messages  EXCEPTION CAUGHT WHILE PROCESSING REQUEST !!! GetEndpointsRequest
Cannot read property 'length' of null
TypeError: Cannot read property 'length' of null
    at OPCUAServer.OPCUABaseServer._on_GetEndpointsRequest (E:\remote_connection
\node_modules\node-opcua\lib\server\base_server.js:269:28)
    at OPCUAServer.OPCUABaseServer.on_request (E:\remote_connection\node_modules
\node-opcua\lib\server\base_server.js:201:21)
    at OPCUAServerEndPoint.<anonymous> (E:\remote_connection\node_modules\node-o
pcua\lib\server\opcua_server.js:249:14)
    at emitThree (events.js:97:13)
    at OPCUAServerEndPoint.emit (events.js:175:7)
    at ServerSecureChannelLayer.<anonymous> (E:\remote_connection\node_modules\n
ode-opcua\lib\server\server_end_point.js:113:14)
    at emitOne (events.js:77:13)
    at ServerSecureChannelLayer.emit (events.js:169:7)
    at ServerSecureChannelLayer._on_common_message (E:\remote_connection\node_mo
dules\node-opcua\lib\server\server_secure_channel_layer.js:894:18)
    at emitMany (events.js:108:13)
    at emit (events.js:182:7)
    at MessageBuilder._decode_message_body (E:\remote_connection\node_modules\no
de-opcua\lib\misc\message_builder.js:443:22)
    at MessageBuilderBase._feed_messageChunk (E:\remote_connection\node_modules\
node-opcua\lib\misc\message_builder_base.js:183:18)
    at null.<anonymous> (E:\remote_connection\node_modules\node-opcua\lib\misc\m
essage_builder_base.js:46:14)
    at emitOne (events.js:77:13)
    at emit (events.js:169:7)
    at PacketAssembler.feed (E:\remote_connection\node_modules\node-opcua\lib\tr
ansport\packet_assembler.js:96:14)
    at MessageBuilderBase.feed (E:\remote_connection\node_modules\node-opcua\lib
\misc\message_builder_base.js:147:30)
    at null.<anonymous> (E:\remote_connection\node_modules\node-opcua\lib\server
\server_secure_channel_layer.js:331:37)
    at emitOne (events.js:77:13)
    at emit (events.js:169:7)
    at _on_message_received (E:\remote_connection\node_modules\node-opcua\lib\tr
ansport\tcp_transport.js:173:14)
    at null.<anonymous> (E:\remote_connection\node_modules\node-opcua\lib\transp
ort\tcp_transport.js:250:30)
    at emitOne (events.js:77:13)
    at emit (events.js:169:7)
    at PacketAssembler.feed (E:\remote_connection\node_modules\node-opcua\lib\tr
ansport\packet_assembler.js:96:14)
    at Socket.<anonymous> (E:\remote_connection\node_modules\node-opcua\lib\tran
sport\tcp_transport.js:257:34)
    at emitOne (events.js:77:13)
    at Socket.emit (events.js:169:7)
    at readableAddChunk (_stream_readable.js:146:16)
    at Socket.Readable.push (_stream_readable.js:110:10)
    at TCP.onread (net.js:523:20)
13:55:06:712 0 ServiceFault  status =  BadInternalError (0x80020000)
                Bytes Read :  94
             Bytes Written :  2780
   time to receive request :  0.004  sec
   time to process request :  0.038  sec
   time to send response   :  0.005  sec
 Sequence Header SequenceHeader { sequenceNumber: 3, requestId: 3 }
 Sequence Number =  3
 Received server interruption from user
 shutting down ...
server closed : all connections have ended
 shutting down completed
 done

```

erossignon, 2016-01-06:
Thank you  for spotting this,

gunnarleffler, 2016-01-06:
Thank you for this project! I'd love to pitch in at some point in the future.

erossignon, 2016-01-07:
:+1:  let me know about it. Will you publish some slideware ?

erossignon, 2016-01-07:
@gunnarleffler, let me also now if the problem has been fixed so we can close this issue.

gunnarleffler, 2016-01-07:
@erossignon It's now working! Feel free to close this bug. As far as the slideware, I'll be sure to mention this project in my next relevant presentation.

closed 2016-01-08 by erossignon