node-opcua/node-opcua #155: Certificate error on running simple_server or simple_client

deepakmlvd, 2016-01-07:
Hello,

I followed the steps given in the Getting Started section. I recieve the following error.

AssertionError: fs.existsSync(options.certificateFile)
   at OPCUAClientBase (C:\Users\dma1bmh\node_modules\node-opcua\sample_client\n
de_modules\node-opcua\lib\client\client_base.js:55:5)
   at new OPCUAClient (C:\Users\dma1bmh\node_modules\node-opcua\sample_client\n
de_modules\node-opcua\lib\client\opcua_client.js:65:21)
   at Object.<anonymous> (C:\Users\dma1bmh\node_modules\node-opcua\sample_clien
\sample_client.js:5:14)
   at Module._compile (module.js:435:26)
   at Object.Module._extensions..js (module.js:442:10)
   at Module.load (module.js:356:32)
   at Function.Module._load (module.js:311:12)
   at Function.Module.runMain (module.js:467:10)
   at startup (node.js:136:18)

Thanks,
Deepak

erossignon, 2016-01-07:
Don't forget to run this command that will generate the default certificates ( they seem to be missing, on your end) 
```sh
npm install
```

deepakmlvd, 2016-01-07:
Thank you. I setup again with following commands and I end up with some other error

cd D:\temp

mkdir test_nodeopcua

cd test_nodeopcua 

npm install node-opcua

git clone https://github.com/node-opcua/node-opcua.git
cd node-opcua
npm init

npm install 

node bin/simple_server

And I end up with the following.

Error: Cannot find module '_generated_/_auto_generated_DiagnosticInfo'
    at Function.Module._resolveFilename (module.js:337:15)
    at Function.Module._load (module.js:287:25)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Object.<anonymous> (D:\temp\test_nodeopc\node-opcua\lib\datamodel\d
tic_info.js:3:26)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Object.<anonymous> (D:\temp\test_nodeopc\node-opcua\lib\datamodel\s
res.js:9:1)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Object.<anonymous> (D:\temp\test_nodeopc\node-opcua\index.js:6:35)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Object.<anonymous> (D:\temp\test_nodeopc\node-opcua\bin\simple_serv
16:13)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:311:12)
    at Function.Module.runMain (module.js:467:10)
    at startup (node.js:136:18)
    at node.js:963:3

deepakmlvd, 2016-01-07:
npm install fails as follows

npm WARN addRemoteGit Error: Command failed: git -c core.longpaths=true co
-get remote.origin.url
npm WARN addRemoteGit
npm WARN addRemoteGit     at ChildProcess.exithandler (child_process.js:20
npm WARN addRemoteGit     at emitTwo (events.js:87:13)
npm WARN addRemoteGit     at ChildProcess.emit (events.js:172:7)
npm WARN addRemoteGit     at maybeClose (internal/child_process.js:818:16)
npm WARN addRemoteGit     at Process.ChildProcess._handle.onexit (internal
_process.js:211:5)
npm WARN addRemoteGit  git://github.com/dtao/lazy.js.git resetting remote
rs\dma1bmh\AppData\Roaming\npm-cache\_git-remotes\git-github-com-dtao-lazy
t-e712e9dc because of error: { [Error: Command failed: git -c core.longpat
e config --get remote.origin.url
npm WARN addRemoteGit ]
npm WARN addRemoteGit   killed: false,
npm WARN addRemoteGit   code: 1,
npm WARN addRemoteGit   signal: null,
npm WARN addRemoteGit   cmd: 'git -c core.longpaths=true config --get remo
gin.url' }
npm WARN addRemoteGit Error: Command failed: git -c core.longpaths=true co
-get remote.origin.url
npm WARN addRemoteGit
npm WARN addRemoteGit     at ChildProcess.exithandler (child_process.js:20
npm WARN addRemoteGit     at emitTwo (events.js:87:13)
npm WARN addRemoteGit     at ChildProcess.emit (events.js:172:7)
npm WARN addRemoteGit     at maybeClose (internal/child_process.js:818:16)
npm WARN addRemoteGit     at Socket.<anonymous> (internal/child_process.js
1)
npm WARN addRemoteGit     at emitOne (events.js:77:13)
npm WARN addRemoteGit     at Socket.emit (events.js:169:7)
npm WARN addRemoteGit     at Pipe._onclose (net.js:469:12)
npm WARN addRemoteGit  git://github.com/dtao/doctrine.git#line-numbers-opt
setting remote C:\Users\dma1bmh\AppData\Roaming\npm-cache\_git-remotes\git
b-com-dtao-doctrine-git-a87d5d22 because of error: { [Error: Command faile
 -c core.longpaths=true config --get remote.origin.url
npm WARN addRemoteGit ]
npm WARN addRemoteGit   killed: false,
npm WARN addRemoteGit   code: 1,
npm WARN addRemoteGit   signal: null,
npm WARN addRemoteGit   cmd: 'git -c core.longpaths=true config --get remo
gin.url' }
npm ERR! git clone --template=C:\Users\dma1bmh\AppData\Roaming\npm-cache\_
motes\_templates --mirror git://github.com/dtao/lazy.js.git C:\Users\dma1b
Data\Roaming\npm-cache\_git-remotes\git-github-com-dtao-lazy-js-git-e712e9
oning into bare repository 'C:\Users\dma1bmh\AppData\Roaming\npm-cache\_gi
tes\git-github-com-dtao-lazy-js-git-e712e9dc'...
npm ERR! git clone --template=C:\Users\dma1bmh\AppData\Roaming\npm-cache\_
motes\_templates --mirror git://github.com/dtao/lazy.js.git C:\Users\dma1b
Data\Roaming\npm-cache\_git-remotes\git-github-com-dtao-lazy-js-git-e712e9
tal: Unable to look up github.com (port 9418) (No such host is known. )
npm ERR! git clone --template=C:\Users\dma1bmh\AppData\Roaming\npm-cache\_
motes\_templates --mirror git://github.com/dtao/doctrine.git C:\Users\dma1
pData\Roaming\npm-cache\_git-remotes\git-github-com-dtao-doctrine-git-a87d
Cloning into bare repository 'C:\Users\dma1bmh\AppData\Roaming\npm-cache\_
motes\git-github-com-dtao-doctrine-git-a87d5d22'...
npm ERR! git clone --template=C:\Users\dma1bmh\AppData\Roaming\npm-cache\_
motes\_templates --mirror git://github.com/dtao/doctrine.git C:\Users\dma1
pData\Roaming\npm-cache\_git-remotes\git-github-com-dtao-doctrine-git-a87d
fatal: Unable to look up github.com (port 9418) (No such host is known. )
npm ERR! Windows_NT 6.1.7601
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\no
node_modules\\npm\\bin\\npm-cli.js" "install"
npm ERR! node v4.2.3
npm ERR! npm  v2.14.7
npm ERR! code 128

npm ERR! Command failed: git -c core.longpaths=true clone --template=C:\Us
a1bmh\AppData\Roaming\npm-cache\_git-remotes\_templates --mirror git://git
m/dtao/lazy.js.git C:\Users\dma1bmh\AppData\Roaming\npm-cache\_git-remotes
ithub-com-dtao-lazy-js-git-e712e9dc
npm ERR! Cloning into bare repository 'C:\Users\dma1bmh\AppData\Roaming\np
e\_git-remotes\git-github-com-dtao-lazy-js-git-e712e9dc'...
npm ERR! fatal: Unable to look up github.com (port 9418) (No such host is
 )
npm ERR!
npm ERR!
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     D:\temp\test_nodeopc\node_modules\node-opcua\npm-debug.log

erossignon, 2016-01-07:
Wierd: 
I have done this again on a new folder and it it work like a charm:

``` 
c:>mkdir foo
c:>cd foo
c:\foo>git clone https://github.com/node-opcua/node-opcua
c:\foo>cd node-opcua
c:\foo\node-opcua>cd node-opcua
c:\foo\node-opcua>npm install
```
I am using node version 5.3.0 and npm 3.3.12
I have noticed a few gotcha with the new npm module layout, it would be worth trying to delete the node_modules folder and rerun ```npm i``` or alternatively  redo the all install on a empty folder again.

Let me know

erossignon, 2016-01-07:
While I was there I made a few adjustments to make sure the demo command
```node bin/simple_client.js -e "opc.tcp://opcserver.mAutomation.net:4841" -n="ns=1;s=EVR2.system.RTC_SEC"```` handles gracefully the non-conformant DataChange notification.

deepakmlvd, 2016-01-08:
I followed the steps on a new folder (node version 5.4.0 and npm 3,3.12. i followed several steps given in http://stackoverflow.com/questions/31744852/npm-install-giving-error-while-accessing-git-url

but it did not help. Can I get the offline certificate for now? I get certificate error when I get run sample_server. 

Thanks,
Deepak

erossignon, 2016-01-08:
It looks like you are working behind a corporate proxy that prevent you accessing the required ressource.
On windows, the openssl package is downloaded and and installed in the node-opcua/bin folder during setup. The certificates are created by this instance of openssl.
Can you verify that you have  a 'node-opcua/bin\openssl' folder ?
Can you check the content of the 'node-opcua\certificate' folder ?
What is your flavor of windows ? win32 ? win64 ? Intel ? Arm ?

Certificates can be created also by the following command ( in verbose mode )
```
C:foo\node-opuca>node bin\crypto_create_CA.js -dev
```
What is the outcome of this command in your side ?

deepakmlvd, 2016-01-08:
Yes I am behind a corporate proxy.

Can you verify that you have a 'node-opcua/bin\openssl' folder ?
No, the folder does not exist

Can you check the content of the 'node-opcua\certificate' folder ?
The contents of the folder is readme.txt

What is your flavor of windows ? win32 ? win64 ? Intel ? Arm ?
Windows 7.64-Bit. Intel i5

C:foo\node-opuca>node bin\crypto_create_CA.js -dev

          set OPENSSL_CONF =C:/Users/dma1bmh/node_modules/node-opcua/certificate
s/PKI/CA/conf/caconfig.cnf
checking presence of  C:\Users\dma1bmh\node_modules\node-opcua\bin\openssl\opens
sl.exe
openssl seems to be missing and need to be installed
 process.env.PROCESSOR_ARCHITEW6432  = undefined
downloading http://indy.fulgan.com/SSL/openssl-1.0.2d-i386-win32.zip
{ [Error: getaddrinfo ENOTFOUND indy.fulgan.com indy.fulgan.com:80]
  code: 'ENOTFOUND',
  errno: 'ENOTFOUND',
  syscall: 'getaddrinfo',
  hostname: 'indy.fulgan.com',
  host: 'indy.fulgan.com',
  port: 80 }

erossignon, 2016-01-08:
this is the isse: 
your proxy prevents you to access the openssl package in  http://indy.fulgan.com/SSL/openssl-1.0.2d-i386-win32.zip
you can try to download it on a different computer and unzip it in the node-opcua\bin\openssl folder

deepakmlvd, 2016-01-11:
Yes I will try again at home on my personal laptop.

sanjaymishraGit, 2017-02-28:
Hallo Erossignon ,

I am getting error" AssertionError: fs.existsSync(this.certificateFile), "Certificate file must exist :" + this.certificateFile" . Here are some details:
1, node JS version is v6.10.0
2.Can you verify that you have a 'node-opcua/bin\openssl' folder ? : "NO"
3. Can you check the content of the 'node-opcua\certificate' folder ? : Yes
4. What is your flavor of windows ? win32 ? win64 ? Intel ? Arm ?Windows 7.64-Bit. Intel i5

any help here plz
Thanks

erossignon, 2017-06-14:
Once  OpenSSL has been manually installed, you need to rerun the utility that creates the certificates. 
```
node bin/crypto_create_CA.js demo --dev 
```

closed 2017-06-14 by erossignon