node-opcua/node-opcua #156: sampling func shall not be re-entrant

danners, 2016-01-14:
Hello,
i am trying to monitor a variable with a refresh period of 1s. Sometimes the reading of the variable takes longer then that. (i need to wait until i get an answer from an external device).
Because of that i get the error:

`
 AssertionError: !self._is_sampling, "sampling func shall not be re-entrant !! fix it"
    at MonitoredItem._on_sampling_timer (/root/opc/node_modules/node-opcua/lib/server/monitored_item.js:332:9)
`

Im not an OPC expert so the first question is, if sampling faster then the value can be read is forbidden by OPC.
Or are there techniques i could use for such a case to avoid that error?

erossignon, 2016-01-18:
Simon, you found a case that was not properly tested. I have added a unit test to reproduce the situation you are describing and provided a fix. 
( note Sampling interval is explained in  OPCUA Release 1.0.3 Part 4  5.12.1.2 Sampling interval.)
Let me know if it works as expected on your end.

danners, 2016-01-19:
Hello,
with the patch it now works as expected.
Thanks a lot.

closed 2016-01-19 by danners