node-opcua/node-opcua #157: Writing to a node with Write access capabilities

riarashid, 2016-01-19:
I am trying to write to a node which has write access capabilities. The datatype of the node is Binary. But getting an error: 

MessageBuilder : ERROR DETECTED IN event handler
[TypeError: Cannot assign to read only property 'dataType' of 1]

My code is copied below 

    function _write_node_value(session,nodeId,value,attributeId,expectedStatusCodes,callback) {

    	var nodesToWrite = [
    	
            {
                nodeId: "ns=6;s=::AsGlobalPV:opcuatest1",
                attributeId: opcua.AttributeIds.Value,
                indexRange: null,
                value: {
                    dataType: opcua.DataType.Binary,
                    value: 1
              }
            }
        ];

        the_session.write(nodesToWrite, function (err, statusCodes) {
            if (!err) {
                statusCodes.length.should.equal(nodesToWrite.length);
                statusCodes[0].should.eql(opcua.StatusCodes.BadNotWritable);
            }
            done(err);
        });

    },

Could you help me with finding the error in the above code?

erossignon, 2016-01-19:
```opcua.DataType.Binary``` is not a valid enum value for ```DataType`` .
Please check : https://github.com/node-opcua/node-opcua/blob/master/schemas/DataType_enum.js
May be you mean ```opcua.DataType.Boolean``` or  ```opcua.DataType.ByteString``` ?

riarashid, 2016-01-25:
Thnak you for pointing that out.

What i need is opcua.Datatype.Boolean. I tried with that too. Still it is showing the same error. Do you have any more suggestions for resolving this? Boolean datatype can have the two values 0 and 1? i am trying to assgn the true value ie. 1 here.
Thank you for your help.

erossignon, 2016-01-25:
Can you provide more information about your context ? 
What is the server you are trying to access ?
do you have a more detailed stack trace ?

riarashid, 2016-02-02:
I tried with the above code. Now I am getting a new error

MessageBuilder : ERROR DETECTED IN event handler
[TypeError: Cannot read property 'equal' of undefined]

What I am trying to do here is to write to a node which has write access capabilities.
Many Thanks for your help

erossignon, 2016-02-02:
Add
'''var should = require("should");''' on top of your file

riarashid, 2016-02-08:
Hello, 

I tried with adding the var should = require("should"); on top of my file. But then I am getting an error "Error: Cannot find module 'should'

Then i tried with commenting the two lines. But still the value of the node is not getting updated.

I am copying my entire code below:

var opcua = require("node-opcua");
var async = require("async");

var DataType = opcua.DataType;
var AttributeIds = opcua.AttributeIds;
var client = new opcua.OPCUAClient();
var endpointUrl = "opc.tcp://131.188.251.98:4841";

var the_session, the_subscription;

async.series([

    // step 1 : connect to
    function(callback)  {
    	client.connect(endpointUrl,function (err) {
    	    if(err) {
    	        console.log(" cannot connect to endpoint :" , endpointUrl );
    	    } else {
    	        console.log("connected !");
    	    }
    	    callback(err);
    	});
    },

    // step 2 : createSession
    function(callback) {
    	client.createSession( function(err,session) {
    	    if(!err) {
    	        the_session = session;
    	    }
    	    callback(err);
    	});
    },

    // step 3 : browse
    function(callback) {
    	the_session.browse("RootFolder", function(err,browse_result){
    	    if(!err) {
    	        browse_result[0].references.forEach(function(reference) {
    	            console.log( reference.browseName.toString());
    	        });
    	    }
    	    callback(err);
    	});
    },

 // step 4' : read a variable with read
     function(callback){
    	the_session.readVariableValue("ns=6;s=::AsGlobalPV:opcuatest2", function(err,dataValue){
    		if (!err) {
    	        console.log(" opcuatest2 = " , dataValue.toString());
    	    }
    	    callback(err);
        });
    },
    
  // step 5: to write value to a node
    
    function _write_node_value(session,nodeId,value,attributeId,expectedStatusCodes,callback) {

        var nodesToWrite = [
            {
                nodeId: "ns=6;s=::AsGlobalPV:opcuatest1",
                attributeId: opcua.AttributeIds.Value,
                indexRange: null,
                value: { /* dataValue*/
                    serverTimestamp: new Date(2015, 5, 2),
                    serverPicoseconds: 20,
                    sourceTimestamp: new Date(2015, 5, 3),
                    sourcePicoseconds: 30,
                    value: { /* Variant */
                        dataType: opcua.DataType.Boolean,
                         value: true
                    }
              }
            }
        ];

        the_session.write(nodesToWrite, function (err, statusCodes) {
            if (!err) {
                //statusCodes.length.should.equal(nodesToWrite.length);
                //statusCodes[0].should.eql(opcua.StatusCodes.BadNotWritable);
            }
            //done(err);
        });

    },
        
      function(callback){
        	the_session.readVariableValue("ns=6;s=::AsGlobalPV:opcuatest2", function(err,dataValue){
        		if (!err) {
        	        console.log(" opcuatest2 = " , dataValue.toString());
        	    }
        	    callback(err);
            });
        },
             
  // close session
    function(callback) {
    	the_session.close(function(err){
    	    if(err) {
    	        console.log("session closed failed ?");
    	    }
    	    callback();
    	});
    }

    ],
    function(err) {
        if (err) {
            console.log(" failure ",err);
        } else {
            console.log("done!");
        }
         client.disconnect(function(){});
    }) ;

Will you kindly help me with this further? Many thanks for your help so far.

erossignon, 2016-02-09:
To get ```` var should = require("should"); ``` working, you need to install the should package by launching the command :
```sh
$ npm install should --save
```
- do not forget to call the ````callback();```  in the write method callback.
 
I suggest also that you check the ```err``` object and the statusCode returned  by the write command as it may be possible that your server doesn't grant you the right to write into this node.

```
        the_session.write(nodesToWrite, function (err, statusCodes) {
            if (!err) {
                console.log("statusCode = ",statusCode.toString());
                //statusCodes.length.should.equal(nodesToWrite.length);
                //statusCodes[0].should.eql(opcua.StatusCodes.BadNotWritable);
            }else {
                console.log("err =",err);
            }
            callback(); // was missing here
        });
```

riarashid, 2016-02-18:
Thank you for your comments. I have made all the changes.

I have updated the write function as:

// to write value to a node
    function _write_node_value(session,nodeId,value,attributeId,expectedStatusCodes,callback) {

        var nodesToWrite = [
            {
                nodeId: "ns=6;s=::AsGlobalPV:opcuatest2",
                attributeId: opcua.AttributeIds.Value,
                indexRange: null,
                value: { /* dataValue*/
                    serverTimestamp: new Date(2015, 5, 2),
                    serverPicoseconds: 20,
                    sourceTimestamp: new Date(2015, 5, 3),
                    sourcePicoseconds: 30,
                    value: { /* Variant */
                        dataType: opcua.DataType.int16,
                         value: 1
                    }
              }
            }
        ];

        the_session.write(nodesToWrite, function (err, statusCodes) {
            if (!err) {
                console.log("statusCode = ",statusCodes[0].toString());
                statusCodes.length.should.equal(nodesToWrite.length);
                statusCodes[0].should.eql(opcua.StatusCodes.BadNotWritable);
            }else {
                console.log("err =",err);
            }
            _write_node_value();
        });

    },

Now I am getting a new error:

statusCode =  BadWriteNotSupported (0x80730000)
MessageBuilder : ERROR DETECTED IN event handler
{ [AssertionError: expected StatusCode {
  description: 'The server not does support writing the combination of value',
  name: 'BadWriteNotSupported',
  value: 2155020288
} to equal StatusCode {
  description: 'The access level does not allow writing to the Node.',
  name: 'BadNotWritable',
  value: 2151350272
} (at value, A has 2155020288 and B has 2151350272)]
  operator: 'to equal',
  expected: 
   StatusCode {
     value: 2151350272,
     description: 'The access level does not allow writing to the Node.',
     name: 'BadNotWritable' },
  details: 'at value, A has 2155020288 and B has 2151350272',
  showDiff: true,
  actual: 
   StatusCode {
     value: 2155020288,
     description: 'The server not does support writing the combination of value',
     name: 'BadWriteNotSupported' },
  stackStartFunction: [Function],
  negate: false,
  assertion: 
   Assertion {
     obj: 
      StatusCode {
        value: 2155020288,
        description: 'The server not does support writing the combination of value',
        name: 'BadWriteNotSupported' },
     anyOne: false,
     negate: false,
     params: 
      { operator: 'to equal',
        expected: [Object],
        message: undefined,
        details: 'at value, A has 2155020288 and B has 2151350272',
        showDiff: true,
        actual: [Object],
        stackStartFunction: [Function],
        negate: false,
        assertion: [Circular] } } }

Does this mean I dont have privilege to write to the node? or Is there something wrong with the data that I am trying to write. Tried accessing the node with UA expert. With that, I am able to write to the node.

Thank you for your valuable help.

erossignon, 2016-02-18:
```dataType: opcua.DataType.int16,``` should be ```dataType: opcua.DataType.Int16```

riarashid, 2016-02-18:
After updating to 'dataType: opcua.DataType.Int16' also, I am getting the same error as above:

statusCode =  BadWriteNotSupported (0x80730000)
MessageBuilder : ERROR DETECTED IN event handler
{ [AssertionError: expected StatusCode {
  description: 'The server not does support writing the combination of value',
  name: 'BadWriteNotSupported',
  value: 2155020288
} to equal StatusCode {
  description: 'The access level does not allow writing to the Node.',
  name: 'BadNotWritable',
  value: 2151350272
} (at value, A has 2155020288 and B has 2151350272)]
  operator: 'to equal',
  expected: 
   StatusCode {
     value: 2151350272,
     description: 'The access level does not allow writing to the Node.',
     name: 'BadNotWritable' },
  details: 'at value, A has 2155020288 and B has 2151350272',
  showDiff: true,
  actual: 
   StatusCode {
     value: 2155020288,
     description: 'The server not does support writing the combination of value',
     name: 'BadWriteNotSupported' },
  stackStartFunction: [Function],
  negate: false,
  assertion: 
   Assertion {
     obj: 
      StatusCode {
        value: 2155020288,
        description: 'The server not does support writing the combination of value',
        name: 'BadWriteNotSupported' },
     anyOne: false,
     negate: false,
     params: 
      { operator: 'to equal',
        expected: [Object],
        message: undefined,
        details: 'at value, A has 2155020288 and B has 2151350272',
        showDiff: true,
        actual: [Object],
        stackStartFunction: [Function],
        negate: false,
        assertion: [Circular] } } }

erossignon, 2016-02-18:
The statusCode returned by the server is ```BadWriteNotSupported``` which seems to tell us that the given node is not writeable. 
It could be also due to a wrong DataType. Are you sure about the DataType of the variable you're trying to access

riarashid, 2016-02-19:
I tried accessing the node with UaExpert. I found the following values
DataType : Int16
AccessLevel : CurrentRead, CurrentWrite
UserAccessLevel : CurrentRead, CurrentWrite

Also i am able to write to the node with UaExpert. I am able to change the value of the node.

Thank you so much for helping me with this. Please do let me know of your suggestions.

riarashid, 2016-03-10:
Hello Erossignon,

Thank you so much for your help so far. I still havent found a way to write to the node. 

The status code returned by the server is still 'BadWriteNotSuppported'.

But from the UAexpert ,it is possible to write to the node and UserAccessLevel is shown as : CurrentRead, CurrentWrite.

Is there any way i can confirm whether I have write privileges to the node from Nodejs? Is there any parameter or certifficate that I can check?

Thank you for your time.

riarashid, 2016-03-16:
Hello,

If the node is not writeable, is there a way to update the UserAcesslevel of the node in nodejs so that the node can be made writeable? How to update the code for that?

Eagerly waiting for your answer. Thank You for your help.

riarashid, 2016-03-16:
Hello, 

If i wanted to access the endpoint URL with Username and password, how can i do that? How can i update the code above so that I can connect to the server with a username and password?

Eagerly waiting for your answer. Thnak you for your help.

erossignon, 2016-03-17:
```var userIdentity = {
            userName: argv.userName,
            password: argv.password
        };
client.createSession(userIdentity, function (err, session) {
            if (!err) {
                the_session = session;
                console.log(" session created".yellow);
                console.log(" sessionId : ", session.sessionId.toString());
            }
            callback(err);
        });
```

riarashid, 2016-03-17:
Hello,
Thank you. When I am trying to create a session with username and password, I am getting the following error.
connected !
 failure  { [AssertionError: serverNonce instanceof Buffer]
  name: 'AssertionError',
  actual: undefined,
  expected: undefined,
  operator: undefined,
  message: 'serverNonce instanceof Buffer',
  generatedMessage: false }
warning : disconnection : closing pending sessions

I tried updating the function _activateSession inside \lib\client\opcua_client.js according to this post 
https://github.com/node-opcua/node-opcua/issues/67
But I am still getiing the same error.

Thnak you so much for your help.

erossignon, 2016-03-17:
OK, Let's get hands on. What OPCUA Server are we talking about ?
Can you expose the OPCUA Server on the internet so I can try to access it remotely ?

erossignon, 2016-04-25:
@riarashid  the remaining issue look similar to #196. can you confirm ?
What is the server that you are trying to connect to ?

pelis, 2016-04-26:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] Writing to a node with Write access capabilities
(#157)" sent on 25.04.2016 22:57:40.

This is the only notification you will receive while this person is away.

riarashid, 2016-04-26:
Hi erossignon,

Yes. This issue does looks similar to #196 

The server we are trying to connect is a opcua server offered by automation studio. Hope the info is sufficient.

pelis, 2016-04-26:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] Writing to a node with Write access capabilities
(#157)" sent on 26.04.2016 11:06:33.

This is the only notification you will receive while this person is away.

antmag, 2016-10-13:
Hey riarashid, hey erossignon,

I work on a dashboard and I can display the opcua server (UA expert) values on my dashboard.
Now, I would like to add a button on my dashboard and when somebody click on this button it call a function. This function will change an OPCUA value. I tried with your  function _write_node_value but I dont know where I have to write this function..
Could you help me please?

Thanks in advance

erossignon, 2016-10-18:
I finally got to the bottom of this one.
Some servers complained with "BadWriteNotSupported" error if an attempt to provide a serverTimestamp in the write.
see https://github.com/OPCFoundation/UA-.NET/blob/master/Stack/Core/Stack/State/NodeState.cs#L3456

closed 2017-09-15 by erossignon

MAHADEVARAJ, 2019-11-05:
> I tried accessing the node with UaExpert. I found the following values
> DataType : Int16
> AccessLevel : CurrentRead, CurrentWrite
> UserAccessLevel : CurrentRead, CurrentWrite
>
> Also i am able to write [...]

please can you share 
how to change value with Uaexpert

JulioRom, 2021-06-30:
hi 
can you try to delete the serverTimestamp attribute in the write, I delete that line, and its work  

`const statusCode = await session.write({
            nodeId: nodeId,
            attributeId: AttributeIds.Value,
            value: {
                statusCode: StatusCodes.Good,
                value: {
                    dataType: DataType.Boolean,
                    value: false
                }
            }
        });`

I hope it helps you