node-opcua/node-opcua #161: Certificates won't generate in v0.0.50 and v0.0.51

Guardiannw, 2016-02-18:
@erossignon My application is crashing everytime I run it using v0.0.50 or v0.0.51 due to a lack of certificates being generated.  I installed the npm packing through npm install and didn't see anything in the output for creation of certificates in 0.0.49, but it looks like you might have silenced that.  I also look in the directory structure, and they do not appear to be in the location that the application is looking for them, if they are there at all.  If I install v0.0.49 and run it, then the application works as expected and the certificates are being generated.  Here is a stack trace of the error, 

```
/home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/better-assert/index.js:37
  throw err;
  ^
AssertionError: fs.existsSync(options.certificateFile)
    at OPCUAClientBase (/home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/lib/client/client_base.js:55:5)
    at new OPCUAClient (/home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/lib/client/opcua_client.js:65:21)
    at new KepwareOpcClient (/home/nwebb/Documents/Clients/ddd/process-control-web/modules/kepware/opcclient.js:19:24)
    at Object.<anonymous> (/home/nwebb/Documents/Clients/ddd/process-control-web/modules/kepware/index.js:13:17)
    at Module._compile (module.js:413:34)
    at Object.Module._extensions..js (module.js:422:10)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Module.require (module.js:367:17)
    at require (internal/module.js:16:19)
```

And here is what is said during the installation of the package, 
```
nwebb@linux-ispc:~/Documents/Clients/ddd/process-control-web> npm install node-opcua@0.0.51 --save

> node-opcua@0.0.51 postinstall /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua
> node postinstall.js

Executing  node bin/generate_opcua_classes.js --clear 
done ... (0)
Executing  node bin/crypto_create_CA.js 
done ... (0)
- node-xml-lite@0.0.3 node_modules/node-xml-lite
process-control-web@0.0.1 /home/nwebb/Documents/Clients/ddd/process-control-web
├── UNMET PEER DEPENDENCY es6-shim@0.34.4
├─┬ node-opcua@0.0.51 
│ ├── async@1.5.0 
│ └── ersatz-node-expat@0.1.6 
├── UNMET PEER DEPENDENCY reflect-metadata@0.1.3
├── UNMET PEER DEPENDENCY rxjs@5.0.0-beta.2
└── UNMET PEER DEPENDENCY zone.js@0.5.15

npm WARN optional Skipping failed optional dependency /chokidar/fsevents:
npm WARN notsup Not compatible with your operating system or architecture: fsevents@1.0.7
npm WARN angular2@2.0.0-beta.6 requires a peer of es6-shim@^0.33.3 but none was installed.
npm WARN angular2@2.0.0-beta.6 requires a peer of reflect-metadata@0.1.2 but none was installed.
npm WARN angular2@2.0.0-beta.6 requires a peer of rxjs@5.0.0-beta.0 but none was installed.
npm WARN angular2@2.0.0-beta.6 requires a peer of zone.js@0.5.14 but none was installed.
npm WARN process-control-web@0.0.1 No repository field.

```

I have replaced the name of our client with the text ddd in the path names for security.

Here is a screenshot of the directory structure after the install:

[image]

I appreciate any help you are able to give on the subject.  Until then, I will revert to v0.0.49.

Thanks,

erossignon, 2016-02-21:
It looks like the certificates haven't been generated during the "npm install" phase.
You should have a lot of .pem and .pub files in the certificate folder

The certifcates are generated in the sequence:
```
Executing  node bin/crypto_create_CA.js 
done ... (0)
```

can you try crypto_create_CA.js in verbose mode so we can analyse why the openssl command has failed.
```
$cd node_modules\node_opcua
$node bin/crypto_create_CA.js
```

Also:
 - what version of node are you using ? ```node --version```
 - what version of openssl is installed on your box ? ```openssl version```
 - what type of linux ? OS ? platform ?

Guardiannw, 2016-02-22:
```
nwebb@linux-ispc:~/Documents/Clients/ddd/process-control-web/node_modules/node-opcua> node bin/crypto_create_CA
          set OPENSSL_CONF =/home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/PKI/CA/conf/caconfig.cnf
        stdout /usr/bin/openssl
CA private key already exists ... skipping
 hostname =  hostname: Name or service not known

Create  Application Certificate for Server & its private key
============================================================ 

 urn =  urn:hostname: Name or service not known:NodeOPCUA-Client
    openssl genrsa -out /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/client_key_1024.pem 1024
        err usage: genrsa [args] [numbits]
        err  -des            encrypt the generated key with DES in cbc mode
        err  -des3           encrypt the generated key with DES in ede cbc mode (168 bit key)
        err  -seed
        err                  encrypt PEM output with cbc seed
        err  -aes128, -aes192, -aes256
        err                  encrypt PEM output with cbc aes
        err  -camellia128, -camellia192, -camellia256
        err                  encrypt PEM output with cbc camellia
        err  -out file       output the key to 'file
        err  -passout arg    output file pass phrase source
        err  -f4             use F4 (0x10001) for the E value
        err  -3              use 3 for the E value
        err  -engine e       use engine e, possibly a hardware device.
        err  -rand file:file:...
        err                  load the file (or the files in the directory) into
        err                  the random number generator
ERROR  Command failed: /bin/sh -c openssl genrsa -out /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/client_key_1024.pem 1024
usage: genrsa [args] [numbits]
 -des            encrypt the generated key with DES in cbc mode
 -des3           encrypt the generated key with DES in ede cbc mode (168 bit key)
 -seed
                 encrypt PEM output with cbc seed
 -aes128, -aes192, -aes256
                 encrypt PEM output with cbc aes
 -camellia128, -camellia192, -camellia256
                 encrypt PEM output with cbc camellia
 -out file       output the key to 'file
 -passout arg    output file pass phrase source
 -f4             use F4 (0x10001) for the E value
 -3              use 3 for the E value
 -engine e       use engine e, possibly a hardware device.
 -rand file:file:...
                 load the file (or the files in the directory) into
                 the random number generator
```

Node Version: 
```
node --version
v5.6.0
```

OpenSSL Version: 
```
openssl version
OpenSSL 1.0.1i-fips 6 Aug 2014
```

OS: OpenSUSE LEAP 42.1 Linux

erossignon, 2016-02-22:
the line : 
''' urn =  urn:hostname: Name or service not known:NodeOPCUA-Client'''  is suspiciuous

what is your hostname ? 
```$ hostname```

Guardiannw, 2016-02-23:
```
hostname
linux-ispc.suse
```

Thanks

erossignon, 2016-02-23:
Thanks, can you try this now
```
$cd node_modules/node-opcua
$node
> require("os").hostname()
'linux-6ik6'
> var fqdn = require("fqdn")
undefined
>fqdn();
'linux-6ik6'
```

erossignon, 2016-02-23:
I would also investigate the following links:
- http://askubuntu.com/questions/185070/why-i-get-hostname-name-or-service-not-known-error/185084
- http://ubuntuforums.org/archive/index.php/t-1467978.html
- https://forums.opensuse.org/showthread.php/477398-hostname-Name-or-service-not-known-!

Let me know you findings

Guardiannw, 2016-02-25:
@erossignon 

```
nwebb@linux-ispc:~/Documents/Clients/ddd/process-control-web> cd node_modules/node-opcua/
nwebb@linux-ispc:~/Documents/Clients/ddd/process-control-web/node_modules/node-opcua> node
> require('os').hostname()
'linux-ispc.suse'
> var fqdn = require('fqdn')
undefined
> fqdn()
'hostname: Name or service not known'
> 

```

I have looked into this more and am still unable to figure out what is going on.  I looked at the articles that you stated and I continue to get the same error for the fqdn; even after attempting to set it in the hosts file

erossignon, 2016-03-06:
@Guardiannw   what would be the output of ``` $ hostname --fqdn``` ?

erossignon, 2016-03-06:
see also http://askubuntu.com/a/219588

Guardiannw, 2016-03-07:
```
nwebb@linux-ispc:~> hostname --fqdn
hostname: Name or service not known
```

I looked at the ubuntu post, but my ip address is ever changing and crossing between networks because it is a development laptop.  That wouldn't really be a good solution.

erossignon, 2016-03-07:
I have made some test on my side and I can get the ```hostname: Name or service not known``` behavior you are experiencing when the hostname contains a "." . your hostname is ```linux-ispc.suse```  and contains a ```.```. This may be the issue.

Guardiannw, 2016-03-10:
@erossignon I renamed my hostname so that it is nwebb-OpenSUSE with no "." in it, but it appears that the problems still persists.  Even at the 
```
hostname --fqdn
hostname: Name or service not known
```
level.

Thank you for your help,

erossignon, 2016-03-10:
Please post your /etc/hostname and /etc/hosts files for investigation.

Guardiannw, 2016-03-10:
[hostname.txt](https://github.com/node-opcua/node-opcua/files/167720/hostname.txt)
[hosts.txt](https://github.com/node-opcua/node-opcua/files/167721/hosts.txt)

erossignon, 2016-03-10:
please add the following line in the ```/etc/hosts``` file:

```
127.0.0.1 nwebb-OpenSUSE
```
just below this line
```
127.0.0.1 localhost
```

and retry 
```$ hostname --fqdn```

Guardiannw, 2016-03-11:
```
nwebb@nwebb-OpenSUSE:~/Documents/Clients/Molding Products/process-control-web> hostname --fqdn
nwebb-OpenSUSE
```

Guardiannw, 2016-03-12:
The problem still persists however and the certificates are not being generated.

erossignon, 2016-03-12:
We are making good progress though. Can you:

* delete the ```node-opcua\certificates``` folder 
* run ```node bin/crypto_create_CA.js --force```  and copy/paste the output here ?

Guardiannw, 2016-03-12:
We are! and thank you for every bit of your help through this process.  I will not be able to get to it today, but I can on monday.

Guardiannw, 2016-03-16:
```
nwebb@nwebb-OpenSUSE:~/Documents/Clients/ddd/process-control-web> cd node_modules/node-opcua/
nwebb@nwebb-OpenSUSE:~/Documents/Clients/ddd/process-control-web/node_modules/node-opcua> rm -R certificates/
nwebb@nwebb-OpenSUSE:~/Documents/Clients/ddd/process-control-web/node_modules/node-opcua> node bin/crypto_create_CA.js --force
          set OPENSSL_CONF =/home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/PKI/CA/conf/caconfig.cnf
        stdout /usr/bin/openssl
 .. constructing  /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/
 .. constructing  /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/PKI
 .. constructing  /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/PKI/CA
 .. constructing  /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/PKI/CA/private
 .. constructing  /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/PKI/CA/public
 .. constructing  /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/PKI/CA/certs
 .. constructing  /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/PKI/CA/crl
 .. constructing  /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/PKI/CA/conf

Create Certificate Authority (CA)
================================= 

 ROOT = /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/PKI/CA

Generate the CA private Key
=========================== 

    openssl genrsa -out private/cakey.pem 2048
        err Generating RSA private key, 2048 bit long modulus
        err ...........+++
        err ..........................+++
        err e is 65537 (0x10001)

Generate a certificate request for the CA key
============================================= 

    openssl req -new -text  -extensions v3_ca -key private/cakey.pem  -out private/cakey.csr  -subj "/C=FR/ST=IDF/L=Paris/O=Fake CA for Test/CN=NodeOPCUA"

Generate CA Certificate (self-signed)
===================================== 

    openssl x509 -req -days 3650  -text  -extensions v3_ca -extfile conf/caconfig.cnf -in private/cakey.csr  -signkey private/cakey.pem  -out public/cacert.pem
        err Signature ok
        err subject=/C=FR/ST=IDF/L=Paris/O=Fake CA for Test/CN=NodeOPCUA
        err Getting Private key
        out Certificate:
        out     Data:
        out         Version: 1 (0x0)
        out         Serial Number: 17432622960220115174 (0xf1ed1e38673d10e6)
        out     Signature Algorithm: NULL
        out         Issuer: C=FR, ST=IDF, L=Paris, O=Fake CA for Test, CN=NodeOPCUA
        out         Validity
        out             Not Before: Mar 16 15:00:28 2016 GMT
        out             Not After : Mar 14 15:00:28 2026 GMT
        out         Subject: C=FR, ST=IDF, L=Paris, O=Fake CA for Test, CN=NodeOPCUA
        out         Subject Public Key Info:
        out             Public Key Algorithm: rsaEncryption
        out                 Public-Key: (2048 bit)
        out                 Modulus:
        out                     00:df:5f:2f:c2:37:4f:31:e3:76:ac:d9:3c:a0:e1:
        out                     39:ea:4a:c9:5f:f7:35:32:7a:95:3d:9c:3d:ff:00:
        out                     b1:0b:2b:9c:1b:80:32:3b:48:7a:91:9a:da:08:38:
        out                     7d:a0:7c:73:2c:c6:5f:13:51:f0:35:74:66:f4:74:
        out                     57:99:b5:14:3b:25:80:07:4d:3a:d5:8d:01:de:35:
        out                     01:84:0a:f5:ac:b8:1e:71:8f:e2:d1:29:b4:20:b6:
        out                     1e:e7:a2:00:fb:71:bf:53:b5:2d:68:a1:c1:2d:00:
        out                     5f:43:e4:d6:f8:9b:4f:cb:03:74:9f:7d:db:81:8f:
        out                     21:73:8e:92:a1:5a:30:bd:08:d5:29:e5:31:f4:fc:
        out                     bd:07:87:aa:be:c8:62:09:2a:df:25:04:b1:28:42:
        out                     af:60:8b:f1:2b:7b:f3:90:fc:be:5e:16:72:1a:cd:
        out                     34:09:2c:07:5d:82:08:00:79:a0:2c:ef:81:fa:fb:
        out                     94:32:12:41:85:5c:60:88:5f:12:bb:dd:7d:7c:4e:
        out                     51:3b:83:0e:04:ed:03:ed:7c:85:ea:16:2b:e2:a1:
        out                     a2:06:d3:ed:0a:25:50:57:ce:d3:46:e7:9d:92:64:
[... 31 lines cut ...]
        err  -engine e       use engine e, possibly a hardware device.
        err  -rand file:file:...
        err                  load the file (or the files in the directory) into
        err                  the random number generator
ERROR  Command failed: /bin/sh -c openssl genrsa -out /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/client_key_1024.pem 1024
usage: genrsa [args] [numbits]
 -des            encrypt the generated key with DES in cbc mode
 -des3           encrypt the generated key with DES in ede cbc mode (168 bit key)
 -seed
                 encrypt PEM output with cbc seed
 -aes128, -aes192, -aes256
                 encrypt PEM output with cbc aes
 -camellia128, -camellia192, -camellia256
                 encrypt PEM output with cbc camellia
 -out file       output the key to 'file
 -passout arg    output file pass phrase source
 -f4             use F4 (0x10001) for the E value
 -3              use 3 for the E value
 -engine e       use engine e, possibly a hardware device.
 -rand file:file:...
                 load the file (or the files in the directory) into
                 the random number generator

nwebb@nwebb-OpenSUSE:~/Documents/Clients/ddd/process-control-web/node_modules/node-opcua>
```

erossignon, 2016-03-16:
Strange, I have specifically installed a VM with the same version of Suse, with no problems.

openssl version
OpenSSL 1.0.1i-fips 6 Aug 2014

What is the output of  the following commands ?
```openssl version ```
``openssl genrsa -out /home/nwebb/Documents/Clients/ddd/process-control-we/node_modules/node-opcua/certificates/client_key_1024.pem 1024```` 
- Have you tried on a brand new virtual machine with Suse or Ubuntu  ? 
- Can you contact me directly by email ?

Guardiannw, 2016-03-16:
```
nwebb@nwebb-OpenSUSE:~/Documents/Clients/ddd/process-control-web> openssl version
OpenSSL 1.0.1i-fips 6 Aug 2014
nwebb@nwebb-OpenSUSE:~/Documents/Clients/ddd/process-control-web> openssl genrsa -out /home/nwebb/Documents/Clients/ddd/process-control-web/node_modules/node-opcua/certificates/client_key_1024.pem 1024
Generating RSA private key, 1024 bit long modulus
....++++++
..........++++++
e is 65537 (0x10001)
```
Yes I can contact you directly.
Thanks,

Guardiannw, 2016-03-16:
Well... I thought I could but I am unable to find your email address.

erossignon, 2016-03-17:
``` 
$ git clone https://github.com/node-opcua/node-opcua.git
$ cd node-opcua
$ git shortlog -s -e | grep Etienne
```

This is strange, when the command is run inside nodejs the openssl genrsa fails, when run outside in a shell it succeeds. 
- What is your environment ? CPU ? Memory ? Physical or Virtual Machine etc ... ?
- Have you conducted test on a different machine ? what are the results ?

Guardiannw, 2016-03-22:
I am running with about 8 gig ram and 2 core cpu inside virtualbox.  I have not conducted the tests on any different machines as of yet.

gbaquedano, 2016-04-11:
This looks related to my recently created issue: https://github.com/node-opcua/node-opcua/issues/191

on your project folder, open /node-modules/node-opcua/bin/install_prerequisite.js and change on lines 49 and 50:

```
var cwd = path.join(__dirname, "./openssl");
var cmd = quote(path.join(cwd, "openssl.exe"));
```

for

```
var cwd = path.join(__dirname, "./openssl");
var cmd = path.join(cwd, "openssl.exe");
```

Inside node-modules/node-opcua/bin/ run
`node crypto_create_CA.js`

Check if that works.

Regards

Guardiannw, 2016-04-11:
Will do whenever I get back to my development machine!  Thanks!

Guardiannw, 2016-04-13:
No, it looks like the quote() was not surrounding the path.join in my instance from what I could tell.

Guardiannw, 2016-04-14:
@erossignon This also is not working in windows.
@gbaquedano That did not solve my problem as that line of code was not in my package version.

Guardiannw, 2016-04-14:
@gbaquedano It did solve _a_ problem that I had when running it on windows though.  But still did not solve this problem.

closed 2018-01-25 by erossignon