node-opcua/node-opcua #163: Varibale does not return to Good State after statusCode was bad

ja-kh, 2016-02-22:
My data provider is setting opcua.StatusCodes.Bad when there is some problem getting a valid value for a variable. When the status returns to Good State for the dataprovider the opcua server answers get requests with an internal server error. 
i did some debugging and found out that this error comes from an assert in module ua_variable.js at function isSameVariant called in _Variable_bind_with_simple_get. 
I hope this helps a little to clearify where my problem is located.

erossignon, 2016-02-24:
What version of node-opcua are you using ? Have you tried with recent version 0.0.52 ?

ja-kh, 2016-02-24:
i was using 0.0.49  0.0.51 and it persists in 0.0.52
requesting the following variable with this get function produces this error for me.
```node
        var variable2 = 10.0;
        server.engine.addressSpace.addVariable({
            componentOf: device,
            nodeId: "ns=1;b=1020FFAA", 
            browseName: "MyVariable2",
            dataType: "Double",    
            value: {
                get: function () {
                	if (variable2 >= 20.0)
                	{
                		variable2 = 10.0;
                		console.log("return bad");
                		return opcua.StatusCodes.Bad;
                	}
                	console.log("return normal");
                	variable2 ++;
                    return new opcua.Variant({dataType: opcua.DataType.Double, value: variable2 });
                },
                set: function (variant) {
                    variable2 = parseFloat(variant.value);
                    return opcua.StatusCodes.Good;
                }
            }
        });
```
following is the trace 
```
return normal
return normal
return bad
return normal
func readValueAsync has failed 
 stack AssertionError:  va instanceof Variant
    at isSameVariant (/home/janadam/workspace/netANALYZER_OPCUA_Server/node_modules/node-opcua/lib/address_space/ua_variable.js:155:5)
    at UAVariable.timestamped_get_func_from__Variable_bind_with_simple_get [as _timestamped_get_func] (/home/janadam/workspace/netANALYZER_OPCUA_Server/node_modules/node-opcua/lib/address_space/ua_variable.js:869:38)
    at UAVariable.async_refresh_func [as refreshFunc] (/home/janadam/workspace/netANALYZER_OPCUA_Server/node_modules/node-opcua/lib/address_space/ua_variable.js:832:30)
    at UAVariable.self.asyncRefresh (/home/janadam/workspace/netANALYZER_OPCUA_Server/node_modules/node-opcua/lib/address_space/ua_variable.js:790:26)
    at UAVariable.readValueAsync (/home/janadam/workspace/netANALYZER_OPCUA_Server/node_modules/node-opcua/lib/address_space/ua_variable.js:1172:14)
    at monitoredItem_read_and_record_value_async (/home/janadam/workspace/netANALYZER_OPCUA_Server/node_modules/node-opcua/lib/server/opcua_server.js:1497:10)
    at MonitoredItem.self [as samplingFunc] (/home/janadam/workspace/netANALYZER_OPCUA_Server/node_modules/node-opcua/lib/server/opcua_server.js:1538:13)
    at MonitoredItem._on_sampling_timer (/home/janadam/workspace/netANALYZER_OPCUA_Server/node_modules/node-opcua/lib/server/monitored_item.js:370:27)
    at null._repeat (/home/janadam/workspace/netANALYZER_OPCUA_Server/node_modules/node-opcua/lib/server/monitored_item.js:690:14)
    at wrapper [as _onTimeout] (timers.js:275:19)
 SAMPLING ERROR => { [AssertionError:  va instanceof Variant]
  name: 'AssertionError',
  actual: undefined,
  expected: undefined,
  operator: undefined,
  message: ' va instanceof Variant',
  generatedMessage: false }
return normal
func readValueAsync has failed 
 stack AssertionError:  va instanceof Variant
```

closed 2016-03-12 by erossignon