node-opcua/node-opcua #165: Node OPC-UA server connected to an Ignition client

bnouhanna, 2016-02-26:
I am very very new to the topic and I wouldn't mind some help and directions. I am trying to use the sample server and connect it to an Ignition client but doesn't  see to work. First on my server I have such error. and also on the client side I have errors as shown in the attached file. 
[image]
[image]
[image]

I tried  to add my variables inside a folder named "Objects" so that Ignition can fetch them, but it doesn't seem to be working. Maybe I am doing it the wrong way. Some help is highly appreciated.

alamsha, 2016-03-05:
Try this in Ignition client:

1. Edit OPC Server Connection --->  Show advanced properties ----> Host Override ---> 192.xxx.xxx.xxx
----------------------------

192.xxx.xxx.xxx is your opc ua server ip.

bnouhanna, 2016-03-08:
Thx for ur reply. what you suggested is exactly what I did hence you can see that the server is connected but once I want to access the variables is throw the exception.

erossignon, 2016-03-08:
Do you mean this exception? 
https://cloud.githubusercontent.com/assets/15689673/13353059/e4c1fab8-dc91-11e5-9b7a-71a69889ccc9.png

Can you get the exception raised by ignition from the log ?

alternatively you could try to run the node-opcua server in debug mode by setting the DEBUG env var to ALL. Beware: this will generate a lot of text in the console.

```
set DEBUG=ALL
node myserver.js
```

alamsha, 2016-03-09:
I have forgotten to activate my Ignition OPC UA licence and hence it was not fully functional after 2 hours trial period even though it works normal with some OPC UA clients like Prosys OPC UA. But it doesn't work with UaExpert OPC UA client after trail period. 

Now i received my new IgnitionOPC-UA CD-Key and activated it. FYI, IgnitionOPC-UA CD-Key is 100% free and you can download and use it.

https://inductiveautomation.com/products/ignitionopc/download

InsideManCulver, 2016-03-10:
I have the exact same problem while connecting Ignition with node-opcua. 
The log says the following:

[image]
Does anyone have any ideas what could be wrong. I also have the connection status "connected" but when trying to open the variables folder the exceptions is thrown.

Thank you already in advance!

InsideManCulver, 2016-03-11:
The problem is within Ignition version. Ignition version 7.8.1 gives the java error. In the version 7.8.2 this bug is fixed so updating to this version fixes the problem.

closed 2016-07-18 by erossignon