node-opcua/node-opcua #166: OPC-client and Webserver possible?

lesche, 2016-02-28:
Is it possible to create a client like in the tutorial with an integrated webserver so that the html-files will be accesable from other devices to monitor all opc-values?

erossignon, 2016-03-05:
Yes, have a look at https://github.com/node-opcua/node-opcua-htmlpanel

alamsha, 2016-03-06:
alam@alamltu2 ~ $ cd node-opcua-htmlpanel
alam@alamltu2 ~/node-opcua-htmlpanel (master=) $ node node_modules/node-opcua/bin/simple_server.js > /dev/null &
[1] 23026
alam@alamltu2 ~/node-opcua-htmlpanel (master=) $ 
/home/alam/node-opcua-htmlpanel/node_modules/better-assert/index.js:37
  throw err;
  ^
AssertionError: fs.existsSync(this.certificateFile), "Certificate file must exist :" + this.certificateFile
    at OPCUAServer.OPCUASecureObject.getCertificateChain (/home/alam/node-opcua-htmlpanel/node_modules/node-opcua/lib/misc/opcua_secure_object.js:61:9)
    at OPCUAServer.OPCUASecureObject.getCertificate (/home/alam/node-opcua-htmlpanel/node_modules/node-opcua/lib/misc/opcua_secure_object.js:48:33)
    at new OPCUAServer (/home/alam/node-opcua-htmlpanel/node_modules/node-opcua/lib/server/opcua_server.js:238:27)
    at Object.<anonymous> (/home/alam/node-opcua-htmlpanel/node_modules/node-opcua/bin/simple_server.js:96:14)
    at Module._compile (module.js:413:34)
    at Object.Module._extensions..js (module.js:422:10)
    at Module.load (module.js:357:32)
    at Function.Module._load (module.js:314:12)
    at Function.Module.runMain (module.js:447:10)
    at startup (node.js:141:18)
    at node.js:933:3

[1]+  Exit 1                  node node_modules/node-opcua/bin/simple_server.js > /dev/null
alam@alamltu2 ~/node-opcua-htmlpanel (master=) $

erossignon, 2016-03-06:
@alamsha, the certificate creation failed during post-install step.
this issue seems simimar to https://github.com/node-opcua/node-opcua/issues/161 which is still under investigation.
You could try to regenerate the certificate with the following command:
```
$ node node_modules\node-opcua\bin\crypto_create_CA
````

closed 2016-03-13 by erossignon