node-opcua/node-opcua #168: Symmetric Encryption

Nestoteles, 2016-03-01:
Hey,

I continued experimenting with your libary. And while browsing in the code I stumbeld upon this

```
function AES_128_CBC_Encrypt() {
}
function AES_128_CBC_Decrypt() {
}
function AES_256_CBC_Encrypt() {
}
function AES_256_CBC_Decrypt() {
}
```

I understand that this is work in progress, but i have a few questions:
1. is there any symmetrical encryption already implented (and I just overlooked it)?
2. is it correct that before establishing a secure channel messages are encrypted asymmetrically, and afterwards symmetrically?  (that's what the image in the UA Spec 4 - Services, OpenSecureChannel suggests)
3. If yes I am wondering if messages are encrypted in the secure chanel right now?

Cheers and keep up the good work,
Nestoteles

erossignon, 2016-03-05:
1. symmetrical encryption is used in secure connection.  Secure connection uses asymmetric encryption first to let the client and server exchange a symmetric key in a encrypted manner, Then the connection uses symmetrical until  the key expired. at this time the client and server negociate a new pair of symmetric encryption keys (using a asymmetric encrypted message) and the process continues.( more info in the OPCUA spec )
2. correct.
3. yes message are encrypted if the secure connection is used.

you can find more info in the unit test here:
https://github.com/node-opcua/node-opcua/tree/master/test/secure_connection

Nestoteles, 2016-03-09:
Thanks for the detailed answer.

closed 2016-03-09 by Nestoteles