node-opcua/node-opcua #169: Feature Request: OPC-UA Angular plugin of client code

timoc, 2016-03-04:
Feature request:

I would like to build a client side/browser based Javascript application that can subscribe to opc-ua endpoints. I use Angular as my client side MVC framework, which out of the box comes with has two way data binding with rest API's[1]. Angular also has a plugin framework to add extra data bindings e.g. with backbone[2] or similar.

The feature request is that - I would like to see an angular module, or just documentation, to allow me to re-use the work you are doing with OPC-UA and javascript in node, also available in a standards compliant web browser.

[1] https://docs.angularjs.org/tutorial/step_04
[2] http://backbonejs.org/

alamsha, 2016-03-04:
I think, a CMS provides the framework for building dynamic web pages. I would like to embed the OPC UA client into Couch CMS and build dynamic web pages. This will open up the opportunity for 1000's of web programmers to build web SCADA for RTUs,  using open source CMS.

timoc, 2016-03-07:
@alamsha from your comment, I'm not sure if you are looking for what i am looking for.

I am looking for something strictly client side, that uses SOAP to subscribe to values so that they can be used directly in visualisation graphs and diagrams (for a good example of a pure client side Javascript application see AdminLTE[1]

Can i suggest, If you want to deliver OPC content with an application like a CMS, you're better off using this project as a straight OPC-REST proxy and plugging your CMS into the REST side.

[1] https://almsaeedstudio.com/preview

gilesbradshaw, 2016-03-17:
I wouldn't go near SOAP it's a nightmare to access from a browser.

I've done a web based batch/scada system in the past (and am currently  re using it on another project).  It didn't have any 'browsing' facilities of OPC data though it just allowed the client to subscribe to OPC data via signalR endpoints I used OPCLabs software (not open source and c# .NET on the server)   this talks about it..  http://gilesbradshaw.github.io/scada.html

So more recently I started playing again and implemented a server again in .NET with rest endpoints that let you actually browse the tree of OPC data.  https://github.com/gilesbradshaw/KepWeb this also uses signalR to let you subscribe to  nodes for changes - again this is using OPCLabs.  It has a react based front end.

Currently I am looking at writing a GraphQL server to expose the whole OPC data model and that's what I have been playing with here https://github.com/gilesbradshaw/uaQl

It seems to me graphQl is a really good fit for OPC UA which is also a graph + with react and relay on the client.  pub sub is coming to graphQL as well for push based live updates  although in the mean time I will probably use something like Faye.  (all v much a work in progress though!)

gilesbradshaw, 2016-03-17:
This also looks pretty interesting btw http://wpcp.net/spec/latest-draft.html

alamsha, 2016-03-18:
I found these 2 opensource LGPL frameworks for building powerful web applications interesting. Pls share your opinion.

1. Create high-productivity enterprise web applications using JavaScript and the power of the SmartClient framework.
http://www.smartclient.com/

2. The GLG Toolkit is a powerful and flexible tool for creating custom HMI systems that can be deployed on a variety of software and hardware platforms, and on the Web. The Toolkit's open data interface makes it simple to integrate dynamic displays with existing data sources, or to develop new HMI frameworks.
http://www.genlogic.com/demos.html

gilesbradshaw, 2016-03-18:
Hi not sure about the first one but from what I can see genlogic is either a java applet embedded in the browser or a png file getting polled every three seconds and displayed by the browser.

That's highly unscalable - take for example 20 browsers watching a tank for a minute during which a valve opens.  If the system was using websockets to push data out to the browser then that would be 50 very small messages (ignoring keep alives which are tiny pings) sent to the browsers and 50 dom updates for the browsers - say each message was 1K (it could be much less) maybe 50K of data in total and the browsers would update immediately the valve changed.  With what genlogic are doing you've got each browser downloading 60K every three seconds - so that's 60K * 20 * 50  - that's 60MB of data going over the network just for 50 browsers to experience one valve change.

alamsha, 2016-03-18:
What's your take on this?

"MQTT+FTP" to build robust IIoT projects on Ignition: 
https://inductiveautomation.com/forum/viewtopic.php?f=41&t=15291&sid=d40d1c23bad0defd136ee68237b06d6d

timoc, 2016-03-18:
@gilesbradshaw Without an OPC:binary protocol adaptor, soap seems the easiest route. 
I am not looking for browsing capablities, i will already know the interface my OPC serve will support., though you're right in that the maturity of the server implementation is a factor, but it does depend on your use case. I am only interested in building simple service status visualizations, based on the server object (that is mandatory in the OPC service no browsing required).

closed 2019-08-03 by stale[bot]