node-opcua/node-opcua #171: User Authorization

Nestoteles, 2016-03-10:
Hey,

is there already a possibility to grant different users, different permissions on the server?
For example user1 cannot access node1, but user2 can access node1.

Cheers

erossignon, 2016-03-10:
This is a really nice to have feature ! Any volunteers to propose an implementation ?

eoursel, 2016-03-10:
The accesLevel and userAccessLevel are defined for variables nodes but not for objects for instance. So we could  restrict read or write access to a variable but it's not a general solution.

Nestoteles, 2016-03-11:
I checked the specs, but unfortunately it does not propose a general solution.
See Part 2 4.8 User Authorization
"OPC UA provides a mechanism to exchange user credentials but does not specify how the
applications use these credentials. Client and Server applications may determine in their own way
what data is accessible and what operations are authorized."

and Part 4 6.1.5 Creating a Session
"The authentication service is a central database that can verify that user token provided by the
Client. This authentication service may also tell the Server which access rights the user has."

hr83, 2016-03-30:
Hi
I had the same need and I have implemented a solution that meets my own recommendations at least. I have a pending pull request #145 which allows you to specify a parameter "browseFilter" when adding a folder. Hopefully Etienne will merge my changes to the master branch.

Meanwhile you can try with my own branch!

Cheers

closed 2017-04-08 by erossignon