node-opcua/node-opcua #173: Form 2 of Browse seems to give incorrect results

gilesbradshaw, 2016-03-14:
I'm using this public opc ua server 'opc.tcp://opcua.demo-this.com:51210/UA/SampleServer'

when I do:
    function(callback) {

      var browseDescription = {
         nodeId: "ns=0;i=84",
         referenceTypeId: "ns=0;i=35",
         browseDirection: 2
      };

       session.browse(browseDescription, function(err, browseResult, diags){
        if(!err) {
            console.log(" with browse description::", JSON.stringify(browseResult, null, '\t'));
        }
    callback(err);
        });
    },

I get these results..

     with browse description:: [
        {
                "statusCode": {
                        "value": 0,
                        "description": "No Error",
                        "name": "Good"
                },
                "references": [
                        {
                                "referenceTypeId": "ns=0;i=0",
                                "isForward": false,
                                "nodeId": "ns=0;i=85",
                                "browseName": {
                                        "namespaceIndex": 0
                                },
                                "displayName": {},
                                "nodeClass": "Unspecified",
                                "typeDefinition": "ns=0;i=0"
                        },
                        {
                                "referenceTypeId": "ns=0;i=0",
                                "isForward": false,
                                "nodeId": "ns=0;i=86",
                                "browseName": {
                                        "namespaceIndex": 0
                                },
                                "displayName": {},
                                "nodeClass": "Unspecified",
                                "typeDefinition": "ns=0;i=0"
                        },
                        {
                                "referenceTypeId": "ns=0;i=0",
                                "isForward": false,
                                "nodeId": "ns=0;i=87",
                                "browseName": {
                                        "namespaceIndex": 0
                                },
                                "displayName": {},
                                "nodeClass": "Unspecified",
                                "typeDefinition": "ns=0;i=0"
                        }
                ]
        }
    ]

however when I do:

    function(callback) {
       session.browse('ns=0;i=84', function(err, browseResult){
        if(!err) {
          console.log('browse:', JSON.stringify(browseResult, null, '\t'));
            
        }
    callback(err);
        });
    },

I get:

    browse: [
        {
                "statusCode": {
                        "value": 0,
                        "description": "No Error",
                        "name": "Good"
                },
                "references": [
                        {
                                "referenceTypeId": "ns=0;i=35",
                                "isForward": true,
                                "nodeId": "ns=0;i=85",
                                "browseName": {
                                        "namespaceIndex": 0,
                                        "name": "Objects"
                                },
                                "displayName": {
                                        "text": "Objects"
                                },
                                "nodeClass": "Object",
                                "typeDefinition": "ns=0;i=61"
                        },
                        {
                                "referenceTypeId": "ns=0;i=35",
                                "isForward": true,
                                "nodeId": "ns=0;i=86",
                                "browseName": {
                                        "namespaceIndex": 0,
                                        "name": "Types"
                                },
                                "displayName": {
                                        "text": "Types"
                                },
                                "nodeClass": "Object",
                                "typeDefinition": "ns=0;i=61"
                        },

(&c...)

I am wondering why the results from the first call would be different and seemingly incorrect

erossignon, 2016-03-14:
In form 2 of session#browse method, you may want to specify ```nodeClassMask```, ```includeSubTypes``` and ```resultMask```. 
```
    var browseDescription = [{
        nodeId: "ns=0;i=84",
        referenceTypeId: opcua.resolveNodeId("HasProperty"),
        browseDirection: opcua.browse_service.BrowseDirection.Forward,
        nodeClassMask: 0,// opcua.browse_service.makeNodeClassMask("Variable"),
        includeSubtypes: true,
        resultMask: 0x3F // opcua.browse_service.makeResultMask("BrowseName | NodeClass")
    }];
```
tip: you could use console.log(browseResults[0].toString()) to get pretty output.

gilesbradshaw, 2016-03-14:
Thanks very much

closed 2016-03-14 by gilesbradshaw