node-opcua/node-opcua #174: Attribute types

gilesbradshaw, 2016-03-14:
I am reading attributes with session.readAllAttributes  obviously some are not supported for the node type I am reading.

However is there anyway I can find out what their type WOULD be IF the current node supported them

I am trying to make a graphQL type that would match any node with its attributes.

Many thanks for you help and sorry for all the questions

erossignon, 2016-03-15:
See OPC-UA Spec part 3 - 5.9 Summary of Attributes of the NodeClasses ( page 39 in v 1.0.3)

erossignon, 2016-03-15:
|Attribute      |Variable|Variable Type|Object|Object Type|Reference Type|DataType|Method|View|
|---------------|--------|-------------|------|-----------|--------------|--------|------|----|
AccessLevel    |M| | | | | | | |
ArrayDimensions|O|O| | | | | | |
BrowseName     |M|M|M|M|M|M|M|M|
ContainsNoLoops| | | | | | | |M|
DataType       |M|M| | | | | | |
Description    |O|O|O|O|O|O|O|O|
DisplayName    |M|M|M|M|M|M|M|M|
EventNotifier  | | | | | | | |M|
Executable     | | | | | | |M| |
Historizing    |M| | | | | | | |
InverseName    | | | | |O| | | |
IsAbstract     | |M| |M|M|M| | |
MinimumSamplingInterval |O| | | | | | | |
NodeClass      |M|M|M|M|M|M|M|M|
NodeId         |M|M|M|M|M|M|M|M|
Symmetric      | | | | |M| | | |
UserAccessLevel|M| | | | | | | |
UserExecutable | | | | | | |M| |
UserWriteMask  |O|O|O|O|O|O|O|O|
Value          |M|O| | | | | | |
ValueRank      |M|M| | | | | | |
WriteMask      |O|O|O|O|O|O|O|O|

gilesbradshaw, 2016-03-15:
Thank you - I've just bought a book on UA - as I can see I need to gem up on this :)

closed 2016-03-15 by gilesbradshaw