node-opcua/node-opcua #176: EventNotifiers

gilesbradshaw, 2016-03-17:
I can see how I can subscribe to a variable for datavalue changes however objects can be eventnotifiers - I'm confused as to how to subscribe to events?

erossignon, 2016-03-17:
The right place to start is by scrutinizing the unit tests : have a look at 
[node-opcua/test/end_to_end/u_test_e2e_SubscriptionUseCase_monitoring_events.js](https://github.com/node-opcua/node-opcua/blob/2de35d690059b503eb6809e96fda9351beee0baa/test/end_to_end/u_test_e2e_SubscriptionUseCase_monitoring_events.js)

gilesbradshaw, 2016-03-17:
Hi I'd be dead keen to help improve documentation - partly to assuage guilt for troubling you with so many questions :)  I did notice sometimes things don't seem to appear at all in the auto generated docs - for example ClientSession.call could you possibly also give a pointer as to why that night be then I could do some PRs to fix things up  best wishes and a great library btw!

erossignon, 2016-03-17:
Well spotted , a [```* @method call```  is missing here](https://github.com/node-opcua/node-opcua/blob/fbb6ab6dc582bae6765662b21dec0910cbe5cafe/lib/client/client_session.js#L849)

closed 2016-12-28 by erossignon