node-opcua/node-opcua #178: client.browse returning incomplete data with 'form2' & 'form4'

gilesbradshaw, 2016-03-18:
browsing with a filter is giving me blank results for referenceTypeId &c for the returned references

I am using opc.tcp://opcua.demo-this.com:51210/UA/SampleServer

so when I do: 

````
var browseDescription = {
              nodeId: 'RootFolder',
              referenceTypeId: 'Organizes',
              browseDirection: opcua.browse_service.BrowseDirection.Both 
            };
            
            cloient.browse([browseDescription], function(err, browseResult){
              if(!err) {
                console.log(JSON.stringify(browseResult, null, '\t'));
````

I am getting:

````
[
        {
                "statusCode": {
                        "value": 0,
                        "description": "No Error",
                        "name": "Good"
                },
                "references": [
                        {
                                "referenceTypeId": "ns=0;i=0",
                                "isForward": false,
                                "nodeId": "ns=0;i=85",
                                "browseName": {
                                        "namespaceIndex": 0
                                },
                                "displayName": {},
                                "nodeClass": "Unspecified",
                                "typeDefinition": "ns=0;i=0"
                        },
                        {
                                "referenceTypeId": "ns=0;i=0",
                                "isForward": false,
                                "nodeId": "ns=0;i=86",
                                "browseName": {
                                        "namespaceIndex": 0
                                },
                                "displayName": {},
                                "nodeClass": "Unspecified",
                                "typeDefinition": "ns=0;i=0"
                        },
                        {
                                "referenceTypeId": "ns=0;i=0",
                                "isForward": false,
                                "nodeId": "ns=0;i=87",
                                "browseName": {
                                        "namespaceIndex": 0
                                },
                                "displayName": {},
                                "nodeClass": "Unspecified",
                                "typeDefinition": "ns=0;i=0"
                        }
                ]
        }
]

````

however if I do: 

````
client.browse(['RootFolder'], function(err, browseResult){
              if(!err) {
                console.log(JSON.stringify(browseResult, null, '\t'));
            
````

I get:

````
[
        {
                "statusCode": {
                        "value": 0,
                        "description": "No Error",
                        "name": "Good"
                },
                "references": [
                        {
                                "referenceTypeId": "ns=0;i=35",
                                "isForward": true,
                                "nodeId": "ns=0;i=85",
                                "browseName": {
                                        "namespaceIndex": 0,
                                        "name": "Objects"
                                },
                                "displayName": {
                                        "text": "Objects"
                                },
                                "nodeClass": "Object",
                                "typeDefinition": "ns=0;i=61"
                        },
                        {
                                "referenceTypeId": "ns=0;i=35",
                                "isForward": true,
                                "nodeId": "ns=0;i=86",
                                "browseName": {
                                        "namespaceIndex": 0,
                                        "name": "Types"
                                },
                                "displayName": {
                                        "text": "Types"
                                },
                                "nodeClass": "Object",
                                "typeDefinition": "ns=0;i=61"
                        },
                        {
                                "referenceTypeId": "ns=0;i=35",
                                "isForward": true,
                                "nodeId": "ns=0;i=87",
                                "browseName": {
                                        "namespaceIndex": 0,
                                        "name": "Views"
                                },
                                "displayName": {
                                        "text": "Views"
                                },
                                "nodeClass": "Object",
                                "typeDefinition": "ns=0;i=61"
                        },
                        {
                                "referenceTypeId": "ns=0;i=40",
                                "isForward": true,
                                "nodeId": "ns=0;i=61",
                                "browseName": {
                                        "namespaceIndex": 0,
                                        "name": "FolderType"
                                },
                                "displayName": {
                                        "text": "FolderType"
                                },
                                "nodeClass": "ObjectType",
                                "typeDefinition": "ns=0;i=0"
                        }
                ]
        }
]

````

gilesbradshaw, 2016-03-18:
ah ok got it - i see there are more fields for the browse description- I will add these to my pull request for documentation

erossignon, 2016-03-20:
similar to #173 ?

closed 2016-04-22 by gilesbradshaw

pelis, 2016-04-22:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] client.browse returning incomplete data with
'form2' & 'form4' (#178)" sent on 22.04.2016 09:53:15.

This is the only notification you will receive while this person is away.