node-opcua/node-opcua #181: Needing to ping the server

gilesbradshaw, 2016-03-20:
Once I have created a session I am finding that if I dont keep pinging it for example with:

const pinger = (session)=> {
  session.browse('', function(err, browseResult){
    if(!err){
      setTimeout(()=>pinger(session), 5000);
    }
  });
}

````

then when I do try and use it (to browse) I get: 

````

	"response": {
		"responseHeader": {
			"timeStamp": "2016-03-20T01:33:13.080Z",
			"requestHandle": 45,
			"serviceResult": {
				"value": 2149974016,
				"description": "The session was closed by the client.",
				"name": "BadSessionClosed"
			},
			"serviceDiagnostics": {
				"namespaceUri": -1,
				"symbolicId": -1,
				"locale": -1,
				"localizedText": -1,
				"innerStatusCode": {
					"value": 0,
					"description": "No Error",
					"name": "Good"
				}
			},
			"stringTable": [
				[
					""
				]
			]
		}
	}
}
````

wondering if this is as expected?

gilesbradshaw, 2016-03-20:
ah just saw ba7802314f2c9030f82aa8ca4ef4ef9d93423b06

erossignon, 2016-03-20:
Yes, I was about to point this to you.

closed 2016-04-02 by erossignon