node-opcua/node-opcua #182: graphQL

gilesbradshaw, 2016-03-20:
Hi I thought you might like to see [this](https://ua-ql.herokuapp.com/?query=%7BuaNode(nodeId%3A%22ns%3D2%3Bi%3D10756%22)%20%7B%0A%20%20...%20showNode%0A%20%20references(first%3A10)%20%7B%0A%20%20%20%20edges%20%7B%0A%20%20%20%20%20%20node%20%7B%0A%20%20%20%20%20%20%20%20referenceTypeId%20%7B%0A%20%20%20%20%20%20%20%20%20%20uaNode%20%7B%0A%20%20%20%20%20%20%20%20%20%20%20%20...%20showNode%20%20%0A%20%20%20%20%20%20%20%20%20%20%7D%0A%20%20%20%20%20%20%20%20%7D%0A%20%20%20%20%20%20%20%20typeDefinition%20%7B%0A%20%20%20%20%20%20%20%20%20%20uaNode%20%7B%0A%20%20%20%20%20%20%20%20%20%20%20%20...%20showNode%0A%20%20%20%20%20%20%20%20%20%20%7D%0A%20%20%20%20%20%20%20%20%7D%0A%20%20%20%20%20%20%20%20browseName%20%7B%0A%20%20%20%20%20%20%20%20%20%20name%0A%20%20%20%20%20%20%20%20%7D%0A%20%20%20%20%20%20%20%20uaNode%20%7B%0A%20%20%20%20%20%20%20%20%20%20...%20showNode%0A%20%20%20%20%20%20%20%20%20%20dataType%20%7B%0A%20%20%20%20%20%20%20%20%20%20%20%20...%20expandedNodeIdResult%0A%20%20%20%20%20%20%20%20%20%20%7D%0A%20%20%20%20%20%20%20%20%20%20dataValue%20%7B%0A%20%20%20%20%20%20%20%20%20%20%20%20stringValue%0A%20%20%20%20%20%20%20%20%20%20%20%20typedValue%20%7B%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%20...%20on%20MethodParameterTypedArrayValue%20%7B%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20value%20%7B%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20name%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%7D%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%20%7D%0A%0A%20%20%20%20%20%20%20%20%20%20%20%20%7D%0A%20%20%20%20%20%20%20%20%20%20%7D%0A%20%20%20%20%20%20%20%20%7D%0A%20%20%20%20%20%20%7D%0A%20%20%20%20%7D%0A%20%20%7D%0A%7D%0A%7D%0A%0Afragment%20loctext%20on%20LocalizedTextResult%20%7B%0A%20%20value%20%7B%0A%20%20%20%20value%20%7B%0A%20%20%20%20%20%20text%0A%20%20%20%20%7D%0A%20%20%7D%0A%7D%0A%0Afragment%20showNode%20on%20UANode%20%7B%0A%20%20displayName%20%7B%0A%20%20%20%20...%20loctext%0A%20%20%7D%20%20%0A%7D%0A%0Afragment%20expandedNodeId%20on%20ExpandedNodeId%20%7B%0A%20%20%20%20uaNode%20%7B%0A%20%20%20%20%20%20...%20showNode%20%20%20%20%0A%20%20%20%20%7D%0A%7D%0A%0Afragment%20expandedNodeIdResult%20on%20ExpandedNodeIdResult%20%7B%0A%20%20value%20%7B%0A%20%20%20%20value%20%7B%0A%20%20%20%20%20%20...%20expandedNodeId%0A%20%20%20%20%7D%0A%20%20%7D%20%20%0A%7D "demo") it's using your software to serve opcua data model as graphQL.  Not at all complete yet..  & for some reason IE doesn't like it.

repository is here https://github.com/gilesbradshaw/uaQL

BLamy, 2016-06-15:
I'm actually also very interested in using GraphQL as a facade for OPC.

gilesbradshaw, 2016-06-16:
I think it's a great fit + even more so when graphQL gets subscriptions

gilesbradshaw, 2017-01-15:
@BLamy I got a little further forward with GraphQL and node-opcua - this is now using subscriptions for the live data..  https://gilesbradshaw.github.io/apollo-client/#/

BLamy, 2017-04-19:
@gilesbradshaw Have you tried to mirror OPCUA's SOAP API to graphQL?

https://github.com/OPCFoundation/UA-.NET/blob/master/Stack/Core/Schema/Opc.Ua.Services.wsdl

[image]

gilesbradshaw, 2017-04-20:
Hi - no sorry but I'm sure it would be possible..

closed 2019-08-03 by stale[bot]