node-opcua/node-opcua #184: Subscription.Monitor(): datachangetrigger with deadband on the timestamp

coussej, 2016-03-23:
I would like to be able to specify a parameter to subscription.monitor() to have a to ask the subscription to publish the value anyway if the value has not changed for a specified time. I read the documentation and figured out it might be possible like below, having a simpling interval of 1s and publisch the value a second time if it hasn't changed for 10s. However, I get a **DataChangeTrigger is not defined** error. Is this the right approach to achieve this?

Thanks!

Something like:
```javascript
// install monitored item
      var monitoredItem = the_subscription.monitor({
          nodeId: opcua.resolveNodeId("ns=1,i=18520"),
          attributeId: opcua.AttributeIds.Value
        }, {
          samplingInterval: 1000,
          filter: new opcua.DataChangeFilter({
            trigger: DataChangeTrigger.StatusValueTimestamp,
            deadbandType: DeadBandType.Absolute,
            deadbandValue: 10000
          }),
          discardOldest: true,
          queueSize: 10
        },
        opcua.read_service.TimestampsToReturn.Both
      );
```

erossignon, 2016-03-23:
try ```opcua.subscription_service.DataChangeFilter``` instead.

coussej, 2016-03-24:
OK, now using this:

erossignon, 2016-03-24:
What is the server ?

coussej, 2016-03-24:
I do get an initial value, but I expect to receive it again after the deadbandValue on the timestamp. 

I use the public demo server op OPC Labs: http://www.opclabs.com/resources/product-information/articles/1094-public-demo-opc-xml-da-server-2

erossignon, 2016-03-24:
try to capture the "err" event on the monitored item , this may provide reason for failure
```
       monitoredItem.on("err", function (err_message) {
            console.log(monitoredItem.itemToMonitor.nodeId.toString(), " ERROR".red, err_message);
       });
```

coussej, 2016-03-24:
Hi Etienne, 

The node ID was replaced manually in the issue description and I made a type. My node ID is correct. Here is my code: https://gist.github.com/coussej/49205e23c4d764d0380f
I tried what you said, but no result.

On a side note: when the subscription is installed, we specify `requestedLifetimeCount: 10`, what does this actually do? My monitoring runs for hours without interruption, I would expect that I have to increase this parameter to achieve this?

erossignon, 2016-03-24:
I have try your script.
In fact, everything looks good to me. the node with id ns=2;i=10289 is a static variable in this server, which mean that the value does'nt change unless you write to it. So getting a single data change event is the expected behavior.

coussej, 2016-03-26:
I didn't read it in the documentation, but I was comparing it to a function in the GE Proficy Historian. You can define that a value should only be collected when it's value has exceeded the limit of a deadband (absolute or percentage). There, you can also set that if the value hasn't changed for a certain time (the timestamp deadband), it is collected again anyway. This is especially useful for trending afterwards, because graphs get all messed up if you don't get any values for a longer period of time. As de datachange filters provided by OPCUA looked the same way, I was under the impression it would work in a similar fashion, but I guess the only option is then to collect the value with readValue() after the deadband has elapsed?

erossignon, 2016-03-26:
Yes, OPCUA implementation  minimizes the traffic. Strictly speaking this is an unnecessary update.  You may implement something on your end that provide the last known value on a regular basis regardless of how long ago you received the last notification. No notification means that value hasn't changed.

coussej, 2016-03-27:
Ok. I'll implement something from the client side. Thanks for your extensive help Etienne, really appreciated ! :+1:

closed 2016-03-27 by erossignon