node-opcua/node-opcua #185: readHistoryValue dataValues

mepdr, 2016-03-23:
Hi, have a look at the following example
```
tags = ["ns=2;s=tag1", "ns=2;s=tag2"]

the_session.readHistoryValue(tags, "2015-06-10T09:00:00.000Z", "2016-03-23T23:59:00.000Z", function(err, dataValues, diagnostics) {
    console.log(dataValues[0].historyData.dataValues[0].sourceTimestamp, dataValues[0].historyData.dataValues[0].value.value);
    console.log(dataValues[1].historyData.dataValues[0].sourceTimestamp, dataValues[1].historyData.dataValues[0].value.value);
});
the_session.readHistoryValue(tags.reverse(), "2015-06-10T09:00:00.000Z", "2016-03-23T23:59:00.000Z", function(err, dataValues, diagnostics) {
    console.log(dataValues[0].historyData.dataValues[0].sourceTimestamp, dataValues[0].historyData.dataValues[0].value.value);
    console.log(dataValues[1].historyData.dataValues[0].sourceTimestamp, dataValues[1].historyData.dataValues[0].value.value);
});

```
it returns as output
```
Thu Mar 17 2016 15:56:19 GMT+0100 (ora solare Europa occidentale) 4.55361652374268
Thu Mar 17 2016 15:56:19 GMT+0100 (ora solare Europa occidentale) 4.55361652374268

Mon Mar 21 2016 12:00:46 GMT+0100 (ora solare Europa occidentale) 0.021930918097496
Mon Mar 21 2016 12:00:46 GMT+0100 (ora solare Europa occidentale) 0.021930918097496
```

Apparently, dataValues holds only data belonging to the first tag. 
Is it an issue of node-opcua library or it's related to the opcua server ?

erossignon, 2016-03-23:
Let make a code review: see  https://github.com/node-opcua/node-opcua/blob/master/lib/client/client_session.js#L316 for the client readHistoryValue implementation.
 
It could also comes from the server as you guessed. What server ?

mepdr, 2016-03-24:
How can I check if it's the library or the server?
I don't know what server it is either.

erossignon, 2016-03-24:
I have create a small test reproducing the behavior and I have tested it with the [Prosys OPC UA Simulation Server](https://www.prosysopc.com/products/opc-ua-simulation-server/) with the historycal value MyLevel and MySwitch. and it works like a charm. I suspect that the issue comes from the server. 
[Here is the script](https://gist.github.com/erossignon/b471f709a7255c168a52)

mepdr, 2016-03-25:
Thanks for double checking. 
I'll notify the issue to the server developers.

closed 2016-03-25 by mepdr