node-opcua/node-opcua #187: subscribing to executable attribute of a method

gilesbradshaw, 2016-04-02:
I notice that when I subscribe to the executable attribute of a method on the opc labs public server I get an initial notification but no subsequent ones.

So wondering if it's a bug in node-opcua - or its a but in the public server or I should not be getting notifications when it changes..

erossignon, 2016-04-03:
The executable attribute of a method is unlikely to change. It indicates whether or not the connected user can invoke the method ( with a call Request). I would expect this to be the expected behavior.

gilesbradshaw, 2016-04-03:
On the sample app the executable attribute changes with the state of the Boiler - so if it is Running the executable state of the suspend command is true but if suspended it is false...

However another weird thing is that when I execute a method I do get a notification for the executable attribute changing on the method I executed - but not on the other methods which also change.

Eg monitoring both suspend and resume if I suspend I get a notification for suspend chaging but not resume - even though they both do.

However if I execute the method from another client (not even node-opcua) then I get no notifications at all.

I suspect the problem is on their server though - because your code obviously can subscribe to the attribute and it sometimes works..  Or maybe they just shouldn't be using executable attribute like that.  Its quite useful though to have a way of finding if a method is avaiable.

simulation commands for boiler are here ns=5;i=31  also https://ua-ql.herokuapp.com/ns=5;i=31

erossignon, 2016-04-28:
OK ! I have noticed the behavior that you describe: when the Start Method on the boiler Simulator is called the Excecutable attribute of the Halt method goes from false to true. 
We need to mend a unit test to capture and reproduce the behavior.

gilesbradshaw, 2016-04-29:
Hi do you think the problem is the opclabs server not sending events or us not getting them?

erossignon, 2016-04-29:
I guess this is coming from the server, ( but I may suffer from the developer bias ;-) )
I would try to trace the packet, trace can be activated by setting the DEBUG env variable to ALL or TRACE prior launching the nodejs client code.

erossignon, 2016-05-04:
@gilesbradshaw in the latest commit ( see above) I have recreated a server that expose a "boiler" object. this Boiler object has a simulation "ProgramStateMachine" with start , resume, halt, reset methods. The executable flags of the methods are changing based on the CurrentState of the simulation object.
For instance, when  CurrentState is "Halted" , the executable flag of the "Halt" method is false , true overwise.
I have then created a client ( proxy ) that monitor the executable flag changing with node-opcua. 
I can confirm that node-opcua capture the changes appropriately.
The problem comes from the CAS server , may be we shall report the issue to @mpostol

mpostol, 2016-05-04:
Thank you for feedback. I am wondering if you are using on-premise CommServer downloaded from www.commsvr.com or remote access to the publicly available on the Internet server: `opc.tcp://commsvr.com:51234/UA/CAS_UA_Server`. 
Please note that the public @CAS_UA_Server exposes ADI model, but not Boiler model. Additionally it has no simulator behind, so all values are randomly generated. More you can learn from:
http://www.commsvr.com/Products/OPCUA/OPCUAServers/OPCUAADIServer.aspx

gilesbradshaw, 2016-05-04:
Thanks @mpostol actually it was the OPC labs public server I was accessing and thanks for looking at it @erossignon

closed 2016-06-21 by erossignon