node-opcua/node-opcua #188: MessageBuilder : ERROR DETECTED IN event handler - TypeError: Cannot read property 'join' of null

biancode, 2016-04-04:
I got this five times after subscribing - Status-Code is good and is working well - What is the problem?

MessageBuilder : ERROR DETECTED IN event handler
[TypeError: Cannot read property 'join' of null]
TypeError: Cannot read property 'join' of null
    at process_request_callback (/Users/klaus/.npm-global/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:62:83)
    at _on_message_received (/Users/klaus/.npm-global/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:113:5)
    at emitMany (events.js:121:13)
    at emit (events.js:195:7)
    at MessageBuilder._decode_message_body (/Users/klaus/.npm-global/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua/lib/misc/message_builder.js:445:22)
    at MessageBuilderBase._feed_messageChunk (/Users/klaus/.npm-global/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua/lib/misc/message_builder_base.js:186:18)
    at null.<anonymous> (/Users/klaus/.npm-global/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua/lib/misc/message_builder_base.js:46:14)
    at emitOne (events.js:90:13)
    at emit (events.js:182:7)
    at PacketAssembler.feed (/Users/klaus/.npm-global/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua/lib/transport/packet_assembler.js:96:14)
    at MessageBuilderBase.feed (/Users/klaus/.npm-global/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua/lib/misc/message_builder_base.js:150:30)
    at ClientSecureChannelLayer._on_receive_message_chunk (/Users/klaus/.npm-global/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:793:25)
    at null.<anonymous> (/Users/klaus/.npm-global/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:550:18)
    at emitOne (events.js:90:13)
    at emit (events.js:182:7)
    at _on_message_received (/Users/klaus/.npm-global/lib/node_modules/node-red-contrib-opcua/node_modules/node-opcua/lib/transport/tcp_transport.js:173:14)

biancode, 2016-04-04:
Subscription is done as in your example https://github.com/node-opcua/node-opcua-sample/blob/master/simple_client.js  without timeout terminate

tried to subscribe two Items - one subscription per monitored item - how would you do that?

erossignon, 2016-04-05:
The issue has been fixed recently in 416d912d2673be2d41756348c821395a526bb488
May be you could use github master branch until next version is released ?

erossignon, 2016-04-05:
> tried to subscribe two Items - one subscription per monitored item - how would you do that?
* Try to mutualize Subscriptions, two or three subscriptions with different publish interval  ( 100 ms , 1 s , 10s) could be good enough.
* Add many monitored items to one subscription, each monitored item can have different samplingInterval
* choose the hosting subscription by choosing the most appropriated publish interval.

biancode, 2016-04-05:
works with actual github master
Thank you very much!

closed 2016-04-05 by biancode