node-opcua/node-opcua #191: Error finding OpenSSL prerequisite (Windows, node v4)

gbaquedano, 2016-04-11:
Hello all,
I have a system running Windows Server 2012 R2 and Node 4.4.1. As it has to go through a proxy for network connection I copied the openssl folder to its "node_modules/node-opcua/bin" folder and ran crypto_create_CA, for setup simplicity. I found out that install_prerequisite.js fails to find "openssl.exe" probably due to malformed string while adding quotation marks with "quote" function, defined just above paths.

```
C:\Users\gbaquedano\Desktop\sp-opcua\node_modules\node-opcua\bin>node crypto_create_CA.js
          set OPENSSL_CONF =C:/Users/gbaquedano/Desktop/sp-opcua/node_modules/node-opcua/certificates/PKI/CA/conf/caconfig.cnf
checking presence of  "C:\Users\gbaquedano\Desktop\sp-opcua\node_modules\node-opcua\bin\openssl\openssl.exe"
```
Notice how up to here path seems correct. existsSync returned false all the time so I changed it to **access** and found out that the path is wrong:
```
Error: ENOENT: no such file or directory, access 'C:\Users\gbaquedano\Desktop\sp-opcua\node_modules\node-opcua\bin\"C:\Users\gbaquedano\Desktop\sp-**opcua\node_modules\node-opcua\bin\openssl\openssl.exe"'
openssl seems to be missing and need to be installed
 process.env.PROCESSOR_ARCHITEW6432  = undefined
```

Is this a windows specific issue? Removing quote from paths made it work:
```
var cwd = path.join(__dirname, './openssl');
var cmd = path.join(cwd, 'openssl.exe');
```

I can PR this files with exists changed to access, as exists are now deprecated (see https://nodejs.org/dist/latest-v4.x/docs/api/fs.html#fs_fs_exists_path_callback and https://nodejs.org/dist/latest-v5.x/docs/api/fs.html#fs_fs_exists_path_callback)

erossignon, 2016-04-28:
you probably need to put openssl.exe in "node_modules/node-opcua/bin/openssl/" and not in "node_modules/node-opcua/bin"

erossignon, 2017-06-14:
See also #155

closed 2017-06-14 by erossignon