node-opcua/node-opcua #192: Client connect/disconnect callback on server

Bastian35022, 2016-04-13:
Hi there, great project, thanks a lot for sharing this with us!

One question: Is it possible to set up a callback for an OPCUAServer, which is called when a client connects/disconnects? Is it furthermore possible to get information about this client, like the client's IP? Would be great!

closed 2016-04-14 by erossignon

Bastian35022, 2016-04-26:
On spot what i need. Thanks a lot!

pelis, 2016-04-26:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] Client connect/disconnect callback on server
(#192)" sent on 26.04.2016 17:45:41.

This is the only notification you will receive while this person is away.

commit c3294960 (2016-04-14): fixes #192 adding  event handler 'newChannel' and 'closeChannel' on server to detect when channel is created and closed. add channel#remoteAddress and channel#remotePort property.
commit 28cb8dec (2016-04-14): fixes #192 adding  event handler 'newChannel' and 'closeChannel' on server to detect when channel is created and closed. add channel#remoteAddress and channel#remotePort property.
commit 81e1ab6d (2016-04-14): fixes #192 adding  event handler 'newChannel' and 'closeChannel' on server to detect when channel is created and closed. add channel#remoteAddress and channel#remotePort property.