node-opcua/node-opcua #194: JS OPC UA Client in Webbrowser

askdrgui, 2016-04-18:
Hi,

I'd like to build a webpage that directly communicates OPC UA servers using client side java script. Is this possible based on your solution? If so, do you have any hints how to start?

Any help is appreciated.

Thanks and best regards,
Daniel

pelis, 2016-04-18:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message
"[node-opcua/node-opcua] JS OPC UA Client in Webbrowser (#194)" sent on
18.04.2016 08:50:05.

This is the only notification you will receive while this person is away.

erleeman, 2016-04-19:
I think you can not run the ua client in the browser cause it is based upon nodejs. But implement an ua client on your nodejs server and communicate via socket.io should do the trick.

Maybe you can start with node-opcua-htmlpanel:
https://github.com/node-opcua/node-opcua-htmlpanel

closed 2017-06-14 by erossignon

zemirco, 2020-03-09:
Hey,

I made our OPC UA client for the browser open source https://github.com/HBM/opcua. It is work in progress but activating a session already works. Maybe it's an option for you.

Best regards,
Mirco