node-opcua/node-opcua #195: performance with multiple subscriptions

gilesbradshaw, 2016-04-18:
I've been running as a client on a raspberry pi (I think it's a 1)

Obviously with a slow processor you notice any performance issues

I found that when I was monitoring several variables it used 99% cpu.

I've traced the problem to my bad code - I was creating a fresh subscription for each monitored point..

However what seemed a little odd was that I got the 99% even with very long subscription interval and sampling interval so I was wondering if that was indicative of a performance issue.  Not a problem for me now although i guess it would be if I wanted to monitor lots of different servers.   So just pointing out in case cpu load could be improved.

Here is code I used to replicate with...

````

var opcua = require("node-opcua");
var async = require("async");

var client = new opcua.OPCUAClient();

var endpointUrl ='opc.tcp://opcua.demo-this.com:51210/UA/SampleServer';
var the_subscription = null;
var the_session = null;

function testMonitor(id){
    var newSub = new opcua.ClientSubscription(the_session,{
            requestedPublishingInterval: 10000,
            requestedLifetimeCount: 10,
            requestedMaxKeepAliveCount: 2,
            maxNotificationsPerPublish: 10,
            publishingEnabled: true,
            priority: 10
        });
    var monitoredItem  = newSub.monitor({
            nodeId: opcua.resolveNodeId(id),
            attributeId: 13
          //, dataEncoding: { namespaceIndex: 0, name:null }
        },
        { 
            samplingInterval: 10000,
            discardOldest: true,
            queueSize: 10 
        });
        console.log("-------------------------------------");

        // subscription.on("item_added",function(monitoredItem){
        //xx monitoredItem.on("initialized",function(){ });
        //xx monitoredItem.on("terminated",function(value){ });
        

        monitoredItem.on("changed",function(value){
           console.log(id, " New Value = ",value.toString());
        });

}

async.series([

    // step 1 : connect to
    function(callback)  {

        client.connect(endpointUrl,function (err) {

            if(err) {
                console.log(" cannot connect to endpoint :" , endpointUrl );
            } else {
                console.log("connected !");
            }
            callback(err);
        });
    },
    // step 2 : createSession
    function(callback) {
        client.createSession( function(err,session) {
            if(!err) {
                the_session = session;
            }
            callback(err);
        });

    },
   
    // step 5: install a subscription and monitored item
    //
    // -----------------------------------------
    // create subscription
    function(callback) {

        testMonitor("ns=5;i=1");
        testMonitor("ns=5;i=2");
        testMonitor("ns=5;i=3");
        testMonitor("ns=5;i=4");
        testMonitor("ns=5;i=5");
        testMonitor("ns=5;i=6");
        testMonitor("ns=5;i=7");
        testMonitor("ns=5;i=8");
        testMonitor("ns=5;i=9");
        testMonitor("ns=5;i=10");
        testMonitor("ns=5;i=11");
        testMonitor("ns=5;i=12");
        testMonitor("ns=5;i=13");
        testMonitor("ns=5;i=14");
        testMonitor("ns=5;i=15");
        testMonitor("ns=5;i=16");
        testMonitor("ns=5;i=17");
        testMonitor("ns=5;i=18");
        
    },

    // ------------------------------------------------
    // closing session
    //
    function(callback) {
        console.log(" closing session");
        the_session.close(function(err){

            console.log(" session closed");
            callback();
        });
    },

],
    function(err) {
        if (err) {
            console.log(" failure ",err);
        } else {
            console.log("done!")
        }
        client.disconnect(function(){});
    }) ;

````

pelis, 2016-04-18:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message
"[node-opcua/node-opcua] performance with multiple subscriptions (#195)"
sent on 18.04.2016 15:19:54.

This is the only notification you will receive while this person is away.

erossignon, 2016-04-26:
the   ```   // create subscription    function(callback) {} ``` doesn't call the ```callback();```

erossignon, 2016-04-26:
Giles,
please set the DEBUG environement variable to TRACE before riunning the client, this will make node-opcua more verbose about the transaction that happen between the client and the server.
I see that the server is generating a lot of "ServiceFault" responses. I don't understand why yet.

erossignon, 2016-04-26:
Investigating why the server send ServiceFault error , 
```javascript
xxxxx  <<<<<< _on_message_received  33 ServiceFault
{ /*ServiceFault*/
 responseHeader                /* ResponseHeader                   */: {
   timeStamp                   /* UtcTime                          */: 2016-04-26T18:15:25.866Z
   requestHandle               /* IntegerId                        */: 34               0x22
   serviceResult               /* StatusCode                       */: BadTooManyPublishRequests (0x80780000)
   serviceDiagnostics          /* DiagnosticInfo                   */: {
     namespaceUri              /* Int32                            */: -1
     symbolicId                /* Int32                            */: -1
     locale                    /* Int32                            */: -1
     localizedText             /* Int32                            */: -1
     additionalInfo            /* String                           */: null
     innerStatusCode           /* StatusCode                       */: Good (0x00000)
     innerDiagnosticInfo       /* DiagnosticInfo                   */: {
     }
   }
   stringTable                 /* String                        [] */: [ ] (l=0)
   additionalHeader            /* ExtensionObject                  */: null
 }
};
```

gilesbradshaw, 2016-04-26:
I really appreciate you taking a look - thanks

pelis, 2016-04-27:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] performance with multiple subscriptions (#195)"
sent on 26.04.2016 20:12:23.

This is the only notification you will receive while this person is away.

pelis, 2016-04-27:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] performance with multiple subscriptions (#195)"
sent on 26.04.2016 20:16:39.

This is the only notification you will receive while this person is away.

pelis, 2016-04-27:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] performance with multiple subscriptions (#195)"
sent on 26.04.2016 20:05:00.

This is the only notification you will receive while this person is away.

pelis, 2016-04-27:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] performance with multiple subscriptions (#195)"
sent on 26.04.2016 22:17:08.

This is the only notification you will receive while this person is away.

erossignon, 2016-04-27:
if the subscription keep alive interval is greater than the session timeout itself when the keepSessionAlive is not set, the server will dropped the session on its end and reply with the BadSessionClosed status for any subsequent request. 
*bug2*:  the client doesn't handle the BadSessionClosed  gracefully, it should detect that the server has closed the session , raised an event and also terminate the subscription publish engine.

Giles, the workaround is : 
  * set the keepSessionAlive flag when creating the client
```javascript
    var client = new opcua.OPCUAClient({keepSessionAlive: true});
```
  * create only one subscription and add all monitored item to it.

This being said, the client need to be improved:
  * the ClientPublish engine shall detect when the server returns BadSessionClosed error and stop pushing new PublishRequest in this case.
 * the client could also display a warning of some sort in the console if the subscription keep alive interval is greater than the session timeout itself when the keepSessionAlive flag is not set to help the developer understand that the server session may drop the session abruptly.

pelis, 2016-04-27:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] performance with multiple subscriptions (#195)"
sent on 27.04.2016 06:23:12.

This is the only notification you will receive while this person is away.

closed 2016-04-28 by erossignon

prenaux, 2016-08-20:
Sorry to write this here, but it looks like its related:

```
var client = new opcua.OPCUAClient({keepSessionAlive: true});
```

Is there any reason not to use "keepSessionAlive"? It seems that it would "always" be what you'd want.

erossignon, 2016-08-20:
Pierre,
Usually a opc client will create at least on subscription.  A subscription is designed to send keepalive signal periodically if no value changes. In this case, the  ```{keepSessionAlive:`` parameter doesn't have to be set to  ```true```.
The ```{keepSessionAlive: true}``` is a special mechanism that sets a timer to periodically read the status of the server. This is a trick, to keep the session active in case there is no subscriptions. 
Therefore I believe ``keepSessionAlive`` default value has to be ```false````.

prenaux, 2016-08-21:
Ok, thanks for the clarification.

I'm not using any subscription in my client atm - and I noticed that the client was disconnecting without it, which keepSessionAlive resolved.

Is it better to use subscriptions? I'm using readVariableValue & read to read values when needed atm.

commit 913aab60 (2016-04-28): fixes #195 client now handle gracefully BadSessionClosed and BadTooManyPublishRequest ServiceFaults.