node-opcua/node-opcua #196: createSession: "AssertionError: serverNonce instanceof Buffer" using userName and password

thnimacio, 2016-04-19:
I'm trying to connect to a Server using a username and password, connecting anonymous works, but using a username and password results in the following error:
AssertionError: serverNonce instanceof Buffer
{ [AssertionError: serverNonce instanceof Buffer]
  name: 'AssertionError',
  actual: undefined,
  expected: undefined,
  operator: undefined,
  message: 'serverNonce instanceof Buffer',
  generatedMessage: false }

erossignon, 2016-04-19:
Can you provide additional info such as : 
 - version of node-opcua ( 0.0.52. 0.0.54 ? master branch )
 - what is the server you're trying to connect to.
 - stack trace

thnimacio, 2016-04-20:
Sure,
I got node-opcua using npm, regarding to the package.json in the node_modules/node-opcua folder it is version 0.0.52.
The Server is a industrial SPS ( I think it is this one: http://www.br-automation.com/de/produkte/steuerungssysteme/x20-system/ ) with a build in OPC-UA Server. I do not have any further information about the kind of server, however connecting using other tools like UAExpert works fine.
Stack Trace:
AssertionError: serverNonce instanceof Buffer
    at createUserNameIdentityToken (D:\Projekte\OPC-UA\NodeTest\node_modules\node-opcua\lib\client\opcua_client.js:309:5)
    at OPCUAClient.createUserIdentityToken (D:\Projekte\OPC-UA\NodeTest\node_modules\node-opcua\lib\client\opcua_client.js:357:33)
    at OPCUAClient._activateSession (D:\Projekte\OPC-UA\NodeTest\node_modules\node-opcua\lib\client\opcua_client.js:394:10)
    at D:\Projekte\OPC-UA\NodeTest\node_modules\node-opcua\lib\client\opcua_client.js:556:18
    at D:\Projekte\OPC-UA\NodeTest\node_modules\node-opcua\lib\client\opcua_client.js:235:9
    at modified_callback (D:\Projekte\OPC-UA\NodeTest\node_modules\node-opcua\lib\client\client_secure_channel_layer.js:894:24)
    at process_request_callback (D:\Projekte\OPC-UA\NodeTest\node_modules\node-opcua\lib\client\client_secure_channel_layer.js:72:5)
    at _on_message_received (D:\Projekte\OPC-UA\NodeTest\node_modules\node-opcua\lib\client\client_secure_channel_layer.js:113:5)
    at emit (events.js:118:17)
    at MessageBuilder._decode_message_body (D:\Projekte\OPC-UA\NodeTest\node_modules\node-opcua\lib\misc\message_builder.js:445:22)

pelis, 2016-04-20:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] createSession: "AssertionError: serverNonce
instanceof Buffer" using userName and password (#196)" sent on 20.04.2016
08:35:09.

This is the only notification you will receive while this person is away.

erossignon, 2016-04-25:
I suspect the server has a compliance issue. To my understanding serverNonce  shall not be set to null.
the OPCUA Spec says:
> serverNonce ByteString A random number that should never be used in any other request.
**This number shall have a minimum length of 32 bytes.**
The Client shall use this value to prove possession of its Application Instance
Certificate in the ActivateSession request.
This value may also be used to prove possession of the userIdentityToken it
specified in the ActivateSession request.

pelis, 2016-04-26:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] createSession: "AssertionError: serverNonce
instanceof Buffer" using userName and password (#196)" sent on 25.04.2016
22:48:21.

This is the only notification you will receive while this person is away.

riarashid, 2016-04-26:
Hi erossignon, 

Yes. This issue does looks similar to https://github.com/node-opcua/node-opcua/issues/157

The server we are trying to connect is a opcua server offered by automation studio. Hope the info is sufficient.

riarashid, 2016-04-26:
Hi erossignon,

Yes. This issue does looks similar to #157

The server we are trying to connect is a opcua server offered by automation studio. Hope the info is sufficient.

pelis, 2016-04-26:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] createSession: "AssertionError: serverNonce
instanceof Buffer" using userName and password (#196)" sent on 26.04.2016
10:48:03.

This is the only notification you will receive while this person is away.

pelis, 2016-04-26:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] createSession: "AssertionError: serverNonce
instanceof Buffer" using userName and password (#196)" sent on 26.04.2016
11:05:23.

This is the only notification you will receive while this person is away.

gelidus, 2016-07-04:
We just ran into the same problem. Client will connect without problems, but the authentication stage (session) will fail with: 
``

erossignon, 2016-07-04:
Can you provide the reference of the server you are trying to connect to. it will be easier to investigate by testing using the same configuration that exhibits the issue.

allatrue, 2016-09-30:
We also just run exactly the same problem using KEPServerEX v5.19 and username/password authentication.
First aid was to add following line before calling assert in opcua_client.js / function createUserNameIdentityToken
` if (serverNonce == null) serverNonce = new Buffer(0);`

PR #255 (2016-10-28): opcua_client.js - createUserNameIdentityToken
Enhances opcua_client.js - createUserNameIdentityToken method
+ allows username/password authentication with servers without certificate
+ allows username/password authentication with servers not providing serverNonce - resolves #196
closed 2016-10-28 by erossignon