node-opcua/node-opcua #197: Subscription onTerminated not triggered when killing UA server

coussej, 2016-04-21:
Hi, referring to [this line](https://github.com/node-opcua/node-opcua/blob/master/documentation/sample_client.js#L91) in the sample client: when I have a long running subscription and I kill the UA server, the client is not raising the "terminated" event, so I'm unaware that something happened. It this normal behaviour? Thanks.

pelis, 2016-04-21:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message
"[node-opcua/node-opcua] Subscription onTerminated not triggered when
killing UA server (#197)" sent on 21.04.2016 10:34:41.

This is the only notification you will receive while this person is away.

erossignon, 2016-04-23:
@coussej in fact in the recent version, we have implemented this behavior ( as per the specification). 
If the connection between the client and the server is broken (or if the server crashes), the client try to reconnect to the server automatically and attempt to re-establish the connection and preserve existing subscription.

pelis, 2016-04-24:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] Subscription onTerminated not triggered when
killing UA server (#197)" sent on 23.04.2016 18:20:55.

This is the only notification you will receive while this person is away.

coussej, 2016-04-24:
@erossignon Ok, that's a good thing then. However, how do you then know when it fails to reconnect when the server comes back?

I currently start a async.forever loop in which I both monitor some values with a subscription and periodically poll values with session.read(). If one of the two throws an error somewhere, I disconnect everything and restart the whole thing, but from what you say above that is not necessary?

Thanks!

pelis, 2016-04-24:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] Subscription onTerminated not triggered when
killing UA server (#197)" sent on 24.04.2016 11:11:19.

This is the only notification you will receive while this person is away.

pelis, 2016-04-24:
Just make sure, when you disconnect, you are really disconnected (call back is sometimes taking quiet a while). We have been discussing your issue in #144 quiet intense...

coussej, 2016-05-12:
Hello  @erossignon, @pelis , I'd like to come back to this topic. In my scenario, I do the following things:

- Connect a client
- Establish a session
- In parallel: 
   - start monitoring (= periodically execute read requests)
   - start a subscription 

You mention above that the client will automatically reconnect and re-establisch the session and subscription. However, if I kill the server, both of these will be gone. In the latter case, what should I do, and to which event should I react? Should I only restart everything when I receive an error on session level?

pelis, 2016-05-13:
Hi,
this is the way how I have implemented the „server down“ case. (code base 0.52)
Notes:
the client tries to reconnect according to the settings in connectionStrategy.
When maxRetry is reached the “close” event is fired.
You will get the “backoff” event, whenever the client tries to reconnect

If you really need to know when the server is down you may as well register on the socket events (which is not really nice to do) However, if you need that you could:
client._secureChannel._transport._socket.on("close", function () {…}
Note that this could of course change in later release since it “drills” into local properties.

      function createClient() {
         var connectionStrategy = clientConfig.hasOwnProperty("connectionStrategy") ? clientConfig.connectionStrategy : {};
         connectionStrategy.maxRetry = connectionStrategy.maxRetry || 5;
         connectionStrategy.initialDelay = connectionStrategy.initialDelay || 1000;
         connectionStrategy.maxDelay = connectionStrategy.maxDelay || 5000;
         connectionStrategy.randomisationFactor = connectionStrategy.randomisationFactor || 0;
         options.connectionStrategy = connectionStrategy;
         client = new opcua.OPCUAClient(options);
         
         client.on("close", function (err) {
              ...
          }
          client.on("backoff", function (number, delay) {
            // when number > maxRetry close is triggered
            log.debug("Backoff: " + (number + 1) + " of " + this.connectionStrategy.maxRetry + " - delay: " + delay);
            sendErr("ERR_CONNECT", "Backoff: " + (number + 1) + " of " +      this.connectionStrategy.maxRetry + " - delay: " + delay);
         });

pelis, 2016-05-13:
@etienne 
One suiggestion: Would be nice to be able to stop the backoff process 'gracefully'. I am currently doing that in this rather ugly way:

               if (CODE.forceCancelBackoff) {
                   try {
                         if (client._secureChannel.__call) client._secureChannel.__call._cancelBackoff = true;
                    } catch (e) { 
                       consoleExt.red("unable to execute: 'client._secureChannel.__call._cancelBackoff =       true;'" );
       }
               }

erossignon, 2018-10-28:
OPCUAClient now have many event that can be monitored to detect connection lost and reconnection.
Closing

closed 2018-10-28 by erossignon