node-opcua/node-opcua #198: Kepware client doesn't provide sessionName causing node-opcua server to assert

erossignon, 2016-04-22:
@nathanpocock no found a bug when trying to connect Kepware client to node-opcua server
Node-opcua complains that the ```sessionName``` provided by the connecting client is null.
This is unusual but legal , as investigation made by @eoursel shows:

**Release 1.03 OPC Unified Architecture, Part 4    page 24 - CreateSession Parameters **

> sessionName  - String - Human readable string that identifies the Session. The Server makes this name and the sessionId visible in its AddressSpace for diagnostic purposes. The Client should provide a name that is unique for the instance of the Client. **If this parameter is not specified the Server shall assign a value.**

closed 2016-04-22 by erossignon

pelis, 2016-04-22:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message
"[node-opcua/node-opcua] Kepware client doesn't provide sessionName causing
node-opcua server to assert (#198)" sent on 22.04.2016 09:24:08.

This is the only notification you will receive while this person is away.

pelis, 2016-04-22:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] Kepware client doesn't provide sessionName causing
node-opcua server to assert (#198)" sent on 22.04.2016 09:50:04.

This is the only notification you will receive while this person is away.

commit d4477b71 (2016-04-22): fixes #198 , server should handle createSessionRequest with sessionName = null