node-opcua/node-opcua #199: Issue with connecting OPC DA server

ashwathnarayanar, 2016-04-22:
Hi is their any way to connect OPC DA server using node js . Please suggest me how we can connect.

pelis, 2016-04-22:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message
"[node-opcua/node-opcua] Issue with connecting OPC DA server (#199)" sent
on 22.04.2016 11:06:09.

This is the only notification you will receive while this person is away.

erossignon, 2016-04-25:
duplicates #41

closed 2016-04-25 by erossignon

pelis, 2016-04-26:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] Issue with connecting OPC DA server (#199)" sent on
25.04.2016 22:32:06.

This is the only notification you will receive while this person is away.