node-opcua/node-opcua #200: Unable to addEnumerationType to server.engine.addressSpace

legolassid, 2016-04-22:
Hey,

I would like to add an EnumerationType to my addressSpace. It seems addressSpace.addEnumerationType method has not been exposed. is this intentional?
 There are [tests](https://github.com/node-opcua/node-opcua/blob/4f2538d1580d35f463a7b8d66abb5f281ae97828/test/address_space/test_address_space_add_enumeration.js#L56) for the addEnumerationType function, also, the [docs](http://node-opcua.github.io/api_doc/classes/AddressSpace.html) refer to addressSpace.addEnumeration but point to addressSpace.addEnumerationType.

Thanks

erossignon, 2016-04-24:
Yes, I confirm there is still some work to be done in this area.

pelis, 2016-04-24:
I am out of the office until 02.05.2016.

Während meiner Abwesenheit habe ich nur eingeschränkten Mail-Zugriff.
In dringenden Fällen wenden Sie sich bitte an Peter.Oltmanns@pol-it.de

Note: This is an automated response to your message  "Re:
[node-opcua/node-opcua] Unable to addEnumerationType to
server.engine.addressSpace (#200)" sent on 24.04.2016 15:49:57.

This is the only notification you will receive while this person is away.

erossignon, 2016-06-21:
``

closed 2016-06-21 by erossignon