node-opcua/node-opcua #201: What does $dataType.value contain?

Mohammed-El-Nabulsi, 2016-05-06:
I have a little question regarding the data coming back from an OPC Server. We send it over the wire for processing so the inherited methods are not accessible anymore. It looks like so after being recieved by the server for further processing (in JSON): 

    "BrowseName" : {
        "value" : {
            "$dataType" : {
                "key" : "Boolean",
                "value" : 1
            },
            "$arrayType" : {
                "key" : "Scalar",
                "value" : 0
            },
            "value" : false,
            "dimensions" : null
        },
        "statusCode" : {
            "value" : 0,
            "description" : "No Error",
            "name" : "Good"
        },
        "sourceTimestamp" : ISODate("2016-04-28T06:32:00.113Z"),
        "sourcePicoseconds" : 0,
        "serverTimestamp" : ISODate("2016-05-03T17:32:17.346Z"),
        "serverPicoseconds" : 0
    },

I know that the actual value is stored in value.value. But what is contained in value.$dataType.value?

Also is there a way to restore the node opc ua methods for that plain JSON object?

Thanks alot for your help in advance!

Mohammed-El-Nabulsi, 2016-05-06:
Just found the answer to my first question here: 

https://github.com/node-opcua/node-opcua/blob/master/schemas/DataType_enum.js

erossignon, 2016-05-11:
I am not sure you can simply JSONify any of the node-opcua object to pass them to the presentation layer. They are very complex javascript object , with classes, prototypes  and overloaded properties.

We could imagine to reuse the OPCUA transport encode/decode mechanism to pass them through; but this would require significant amount of work to isolate a portion of the node-opcua library and browserify it.

You could also create your own little object to pass through the over side as suggested in this sample application  
https://github.com/node-opcua/node-opcua-htmlpanel/blob/master/app.js#L109

closed 2016-09-06 by erossignon