node-opcua/node-opcua #204: Array of floats

dom-white, 2016-05-10:
Hi, In my opcua server I am trying to send out arrays of floating point values.
If I am creating the following data value to send:
`
`var testArray = new Float32Array([0.0, 1.1, 2.2, 3.3, 4.4, 5.5, 6.6, 7.7, 8.8, 9.9, 10,0, 11.1);

var dataValue = new opcua.DataValue({
     value: new opcua.Variant({
         dataType: opcua.DataType.Float,
         arrayType:  opcua.VariantArrayType.Array,
         value: testArray
          }),
     statusCode: opcua.StatusCodes.Good,
     sourceTimestamp: new Date(result.createdAt)
});`

I am seeing the following warning message despite using a typed array

> 
Warning ! a array containing  Floatelements has been provided as a generic array.
          This is inefficient as every array value will have to be coerced and verified against the expected type
          It is highly recommended that you use a  typed array  Function  instead

I have tried to follow the source code, but cannot quite work out what is happening, or what I am doing wrong.

Any advice would be gratefully received.

erossignon, 2016-05-11:
Could you provide more context of how
  - you create the variable in the address space
  - what you do with ```dataValue```  that generate the issue ?

if you can provide a self-containing minimalist example of how to reproduce the issue as a gist for instance, this would be good.

dom-white, 2016-05-11:
Hi, thanks. I have looked into this further and it is an issue caused by myself trying to work around another problem I had.

The warning does not show if I use a Typed Array AND set the 'dataType' of the 'opcua.Variant' correctly.

e.g. to DataType.Float

In your examples you create new variables using addressSpace.addVariable() and set the dataType property using a string value e.g. "Float"
Then when you create the 'opcua.Variant' that is used to contain the data that is sent out, you use an opcua type for the dataType e.g. DataType.Float. 
I had presumed:
1. that you had to set it as an opcua type here and that you couldn't use a string here, and 
2. that it was an enum, and not an object. So I though it was valid to set it by using 'this.dataType.value' from the context (which is an enum)

As I call addVariable() from within a loop from an object array containing an array of variables of different types, I only know the data type as a string for each variable, this is why I was trying to obtain the data type back out from opcua's context. I have since changed this to using a 'string' value for the variant which appears to work.

e.g. from 
`var value = new Variant({dataType: this.dataType.value, arrayType: opcua.VariantArrayType.Array, value: myArray});`
to
`var value = new Variant({dataType: "Float", arrayType: opcua.VariantArrayType.Array, value: myArray});`

closed 2016-06-09 by erossignon