node-opcua/node-opcua #205: IOPCCommon::SetClientName

kgmoney, 2016-05-12:
Hi,

I've been using your library for about a year for a small project and it's been working great...awesome work!

Quick question...do you support the IOPCCommon::SetClientName method.  I dug around a bit but haven't found anything.  This function just identifies a client to the server so the server (Kepserver in this case) can show a list of connected clients.

Thanks,
Kevin

erossignon, 2016-05-13:
* Client session can be identified on server side by their "sessionName"
* sessionName is set by client in the createSession request [see opcua_client.js#L187](https://github.com/node-opcua/node-opcua/blob/913aab601fb7ca3af3efb12a7a129a85d16d1cb6/lib/client/opcua_client.js#L187)
* at this time node-ocpua provide a generic mechanism to build a sessionName.
* here is how you can retrieve the session name on the client side

```javascript
 client.createSession(function (err, session) {
    console.log(" the session name is ",session.name);
 });

```

erossignon, 2016-05-13:
now you can set clientName , see example in unit test. let me know if this solves your request.

kgmoney, 2016-05-13:
Wow thanks, I'll give it a shot and get back to you!

kgmoney, 2016-05-13:
So I can see the name in session.name as you mention above, but I'm not seeing that value show up in Kepserver.  Is there anything else I should test?

For reference, this is the Kepware knowledgebase item I'm looking at:  https://www.kepware.com/support/knowledge-base/default.asp?solution=/_ui/selfservice/pkb/PublicKnowledgeSolution/d?&id=50140000000SElB

erossignon, 2016-06-21:
Sorry I have no clue , the kepware article refers to the  old OPC ( DCOM based ) not OPCUA apparently.

closed 2016-06-21 by erossignon