node-opcua/node-opcua #208: How do I connect to a DCOM OPC UA server?

prenaux, 2016-05-17:
Sorry if this is obvious but I'm trying to figure out how to connect to a regular DCOM OPC UA server.

Is it possible? It seems that the TCP/IP protocol isnt really the same as the "standard" DCOM servers that I keep seeing everywhere.

For example this:
http://www.matrikonopc.com/products/opc-drivers/opc-simulation-server.aspx

Is simulation server that seem to work (as in their Explorer sees the value I add to the server) but I do not know how to access that server locally. The IP part seems obscure to me since I cant see any of those in the OPC explorer or server.

I tired to use "opc.tcp://" + hostname + ":135/UA/Matrikon.OPC.Simulation.1" (135 is the standard DCOM port) and Matrikon.OPC.Simulation.1 is what is displayed in the Matrikon OPC simulator...

I'm shooting in the dark at this point, any chance you could clarify what are the requirements / limitations, maybe node-opcua cant connect to DCOM OPC server at all?

erossignon, 2016-05-17:
Node-opcua is only implementing the new TCP/IP UA protocol and will not allow you to connect to a server that implements OPC Classic ( based on Microsoft specific COM/DCOM l).
for more info : http://www.ni.com/white-paper/13843/en/

prenaux, 2016-05-18:
Thanks for the info that clarifies things.

closed 2016-05-18 by erossignon