node-opcua/node-opcua #211: SSL can't connect with client-cert authentication

jlamandecap, 2016-06-06:
#### Environment

 - node : 5.5.0
 - node-opcua :0.0.54
 - OPC Server (Codra Solution) configured for client-cert (X509) authentication.

#### Configuration
 - Certificate of my server : ServerCert.der
 - Private Key for my OPC-UA client : ClientCert.pem
  and corresponding certificate : ClientCert.der

__I think the passphrase for the private Key is a missing option. Anyway, for tests, I have set an empty passphrase.__

```
var certificate = fs.readFileSync("ServerCert.der");

var options = {
  "privateKeyFile" : "ClientCert.pem",
  "certificateFile" : "ClientCert.der",
  "serverCertificate" : certificate,
  "securityPolicy" : securityPolicy_m.SecurityPolicy.Basic256,
  "securityMode" : endpoints_service.MessageSecurityMode.SIGN
}
```

#### Troubles :

```
 ERROR RECEIVED FROM SENDER BadCertificateInvalid (0x80120000) Could not verify security on OpenSecureChannel request.
00000000: 45 52 52 46 47 00 00 00 00 00 12 80 37 00 00 00 43 6f 75 6c 64 20 6e 6f 74 20 76 65 72 69 66 79    ERRFG.......7...Could.not.verify
00000020: 20 73 65 63 75 72 69 74 79 20 6f 6e 20 4f 70 65 6e 53 65 63 75 72 65 43 68 61 6e 6e 65 6c 20 72    .security.on.OpenSecureChannel.r
00000040: 65 71 75 65 73 74 2e                                                                               equest.

 message was
undefined
 message was 2: { /*OpenSecureChannelRequest*/
 requestHeader                 /* RequestHeader                    */: {
   authenticationToken         /* NodeId                           */: ns=0;i=0
   timeStamp                   /* UtcTime                          */: 2016-06-06T13:29:22.890Z
   requestHandle               /* IntegerId                        */: 1               0x1
   returnDiagnostics           /* UInt32                           */: 0               0x0
   auditEntryId                /* UAString                         */: null
   timeoutHint                 /* UInt32                           */: 0               0x0
   additionalHeader            /* ExtensionObject                  */: null
 }
 clientProtocolVersion         /* UInt32                           */: 0               0x0
 requestType                   /* SecurityTokenRequestType         */: ISSUE ( 0)
 securityMode                  /* MessageSecurityMode              */: SIGN ( 2)
 clientNonce                   /* ByteString                       */
BUFFER{00000000: e5 df 7f bb 70 3f fe 29 e4 69 17 3f 38 2d ad ee 0f 46 f0 b1 c3 be e8 ea 06 a2 bd ad b9 26 8a c6    e_.;p?~)di.?8--n.Fp1C>hj."=-9&.F
}
 requestedLifetime             /* UInt32                           */: 600000               0x927c0
};
```

I tried to debug but without success.
Am I missing something in the way I configure the client options ?

For information, I succeeded to connect in a Java Client (Prosys) with the same certificates and keys.

Thx

erossignon, 2016-06-06:
It could be a duplicate of #193 , can you try with master branch ?

erossignon, 2016-06-07:
Certificates must be in PEM form not DER form. Paraphrase on Private key  is not supported yet
You can  convert a DER file (.crt .cer .der) to PEM 
```openssl x509 -inform der -in certificate.cer -out certificate.pem.```

jlamandecap, 2016-06-07:
Thank you Etienne.

I tried using the PEM format for each certificate :

```
var certificate = fs.readFileSync("server.pem");

var options = {
  "privateKeyFile" : "client.key.pem",
  "certificateFile" : "client.crt.pem",
  "serverCertificate" : certificate,
```

Without success but a new error message : 
```
 cannot connect to endpoint : opc.tcp://server:9080/UA/MyServer
 failure  { [AssertionError: typeof certificate === "string"]
  name: 'AssertionError',
  actual: undefined,
  expected: undefined,
  operator: undefined,
  message: 'typeof certificate === "string"',
  generatedMessage: false }
```

I'll give a try with the master branch.

erossignon, 2016-06-07:
There is no "serverCertificate" required in the options of a Server.
``` javascript
var options = {
  "privateKeyFile" : "client.key.pem",
  "certificateFile" : "client.crt.pem",
   // NOT NEEDED "serverCertificate" : certificate,
}
```

erossignon, 2016-07-03:
Julien,
- at this time client-cert (X509) authentication is currently not supported.

askdrgui, 2017-09-14:
Dear Etienne,

is it still the case that connecting to an OPC UA Server with client certificates is not supperted in node-opcua client? I have a server that only accepts client connections with client certificates. Connecting with UA Expert works fine.

Any help is appreciated!

Thanks and best regards,
Daniel

erossignon, 2018-06-28:
closing inactive issue

closed 2018-06-28 by erossignon