node-opcua/node-opcua #217: OPC UA Write Value

shrasheed, 2016-06-15:
Using the html dashboard sample provided I was able to get multiple values displaying on the web page. I can write static values from the writefromweb.js file. But when I try to assign a variable to the value field I get an assertion error.  I pasted the error below for reference. From my html file I can see that "globalwrite"variable is being passed through. What would be the proper way to achieve this? Any help would greatly appreciated!

 **$ node writefromweb.js
 connecting to  opc.tcp://192.168.1.111:4840
 session created
subscription started

C:\chatjs\node_modules\node-opcua\node_modules\better-assert\index.js:37
  throw err;
  ^
 AssertionError: value !== undefined
    at coerceVariantType (C:\chatjs\node_modules\node-opcua\lib\datamodel\varian
t_tools.js:55:13)
    at Object.Variant_Schema.construct_hook (C:\chatjs\node_modules\node-opcua\s
chemas\Variant_schema.js:381:29)
    at new Variant (C:\chatjs\node_modules\node-opcua\_generated_\_auto_generate
d_Variant.js:99:22)
    at new DataValue (C:\chatjs\node_modules\node-opcua\_generated_\_auto_genera
ted_DataValue.js:58:36)
    at new WriteValue (C:\chatjs\node_modules\node-opcua\_generated_\_auto_gener
ated_WriteValue.js:76:19)
    at C:\chatjs\node_modules\node-opcua\_generated_\_auto_generated_WriteReques
t.js:64:74
    at Array.map (native)
    at new WriteRequest (C:\chatjs\node_modules\node-opcua\_generated_\_auto_gen
erated_WriteRequest.js:64:50)
    at ClientSession.write (C:\chatjs\node_modules\node-opcua\lib\client\client_
session.js:386:19)
    at C:\chatjs\writefromweb.js:74:15**

Code from writefromweb.js file: 
`var opcua = require("node-opcua");
var async = require("async");
var color = require("colors");

var client = new opcua.OPCUAClient({

});

var the_subscription,the_session;
var userIdentity  = null;
var globalwrite ;

var express = require('express'),
	app = express(),
	server = require('http').createServer(app),
	io = require('socket.io').listen(server); 
	
server.listen(3000); 

var endpointUrl = "opc.tcp://192.168.1.111:4840";

async.series([
    function(callback) {
        console.log(" connecting to ", endpointUrl.cyan.bold);
        client.connect(endpointUrl,callback);
    },
    function(callback) {

        client.createSession(userIdentity,function (err,session){
            if (!err) {
                the_session = session;
                console.log(" session created".yellow);
            }
            callback(err);
        });
    },
    function(callback) {
        the_subscription=new opcua.ClientSubscription(the_session,{
            requestedPublishingInterval: 500,
            requestedMaxKeepAliveCount:  2000,
            requestedLifetimeCount:      6000,
            maxNotificationsPerPublish:  1000,
            publishingEnabled: true,
            priority: 10
        });

       the_subscription.on("started",function(){
            console.log("subscription started");
            callback();

        }).on("keepalive",function(){
            console.log("keepalive");

        }).on("terminated",function(){
            console.log(" TERMINATED ------------------------------>")
        });

    },
	function (callback) {
		        var nodesToWrite = [
            {
                nodeId: "ns=2;s=S7.DATA2ET200.test",
                attributeId: 13,
                indexRange: null,
                value: { 
                    value: { 
                        dataType: opcua.DataType.Int16,
                         value: 20 /*globalwrite*/
                    }
              }
            }
        ];
		the_session.write(nodesToWrite, function(err,statusCode,diagnosticInfo) {
           if (!err) {
               console.log(" write ok" );
               //console.log(diagnosticInfo);
               //console.log(statusCode);
           }
           callback(err);
       });	
       

    }],
function(err) {
    if (!err) {
        startHTTPServer();
		console.log(globalwrite);
    } else {
        // cannot connect to client
        console.log(err);
    }
});

 

function startHTTPServer(){
app.get('/', function(req, res){ 
	res.sendfile(__dirname + '/index.html');
}); 

io.sockets.on('connection', function(socket){
	socket.on('send message', function(data){
		io.sockets.emit('new message', data); 
		//socket.broadcast.emit('new message', data); 
		console.log(data);
		globalwrite = data; 
	}); 
});

} 
`
Code from writedata.html file:
`<html>
<head>
<title> Write test </title>
<style>
		#chat{
		height: 500px; 
		}
</style>
</head>
<body>
	<div id = "chat"></div>
	<form id = "send-message">
		<input size ="35" id = "message"></input>
		<input type="submit"></input>
	</form>

		<script src="http://code.jquery.com/jquery-latest.min.js"></script>
		<script src="/socket.io/socket.io.js"></script>
		<script>
			jQuery(function($){
			var socket = io.connect(); 
			var $messageFrom = $('#send-message'); 
			var $messageBox = $('#message'); 
			var $chat = $('#chat'); 
			
			// Send message from HTML to JS over socket
			$messageFrom.submit(function(e){
				e.preventDefault(); 
				socket.emit('send message', $messageBox.val()); 
				$messageBox.val('');
			});
			
			// Receive 'data'
			socket.on('new message', function(data){
					$chat.append(data + "<br/>"); 
			}); 
			}); 
		</script>
		</body>
</html>
`

erossignon, 2016-06-21:
The problem is in here
```
io.sockets.on('connection', function(socket){
    socket.on('send message', function(data){
        io.sockets.emit('new message', data); 
        //socket.broadcast.emit('new message', data); 
        console.log(data);
        globalwrite = data;
        console.log(globalwrite);
    }); 
});
```
I suspect globalwrite get null here , it should be a Number

Try
globalwrite = data +0;
or
globalwrite = parseInt(data,10);

shrasheed, 2016-06-21:
Etienne, 

Setting globalwrite = parseInt(data,10) did remove the assertion error. But the value of globalwrite is never sent back to ```the_session.write(nodesToWrite, function(err,statusCode,diagnosticInfo) ```

Can you advise what I could be doing wrong? 

Thank you

erossignon, 2016-06-23:
it's because your code is not calling ```the_session.write``` inside ```socket.on('send message', function(data){ ...}```

shrasheed, 2016-06-23:
Etienne,

Appreciate your help in helping me figure this out. I was able to write from the web and update the values on the server after calling the write function as per your suggestion above. 

Thank you.

closed 2016-06-24 by erossignon