node-opcua/node-opcua #218: finding the nodeId of a node by Browse name

UmerIftikhar, 2016-06-20:
Does this functionality has been added?
As it says here:
https://github.com/node-opcua/node-opcua/blob/master/documentation/creating_a_client.md

erossignon, 2016-06-21:
Yes the TranslateBrowsePathsToNodeIds service has been implemented.
Have a [look at the unit test to discover how it can be used](
https://github.com/node-opcua/node-opcua/blob/4f2538d1580d35f463a7b8d66abb5f281ae97828/test/end_to_end/u_test_e2e_translateBrowsePath.js#L41)

prenaux, 2016-08-18:
It doesn't seem to handle spaces correctly.

When I try to resolve: "/Objects/2:MatrikonOPC Simulation Server (DA)" it throws:
```
Error: Malformed relative path  Simulation Server (DA)Error: Malformed relative path  Simulation Server (DA)
 at makeRelativePath (d:\Work\www-ts-opc\node_modules\node-opcua\lib\address_space\make_relative_path.js:127:19)
    at makeBrowsePath (d:\Work\www-ts-opc\node_modules\node-opcua\lib\address_space\make_browse_path.js:19:23)
``

Resolving "/Objects/Server", etc... works as expected.

closed 2016-08-19 by erossignon

prenaux, 2016-08-20:
Why was this closed? was it fixed or moved to another issue?

erossignon, 2016-08-20:
Yep! this has been fixed and tested in commit 801c87175436aca5ebaeaa2c9393fc11d0cca13f . 
I guess Github automatically closed the issue for us because the  the comment in the commit contained "fixes #218 ..." .

prenaux, 2016-08-20:
I can confirm it works, thanks a lot :)

commit 801c8717 (2016-08-19): fixes #218  :now makeBrowsePath takes into account spaces and parenthesis characters.