node-opcua/node-opcua #221: proposed change in Monitered_Item filtering reveals small bug in apply_filter()

awcullen, 2016-06-24:
I would like to have DataChangeTrigger.StatusValue be the default if filter is null. 
So I changed recordValue() of server/Monitored_Item.js as shown.

`
561     //xx if (self.filter) { 
562         if (!apply_filter(self, dataValue)) { 
563             //xx console.log(" Apply filter",self.filter.toString()); 
564             return; 
565         } 
566     //xx } 

`

This does reveal a small bug in apply_filter()
`
 517     if (!self.filter) { 
518         return statusCodeHasChanged(newDataValue, oldDataValue) || 
519             valueHasChanged(self, newDataValue, oldDataValue, DeadbandType.None); //  +'self'
520     } 

`

erossignon, 2016-07-04:
Thank you for spotting the error in line 519, this code gets never called.
However the change you proposed would cause an error if dataValue Variant DataType is not a number the valueHasChanged method requires that the value in a Int a Float or a Double and would assert if the variant type is a String for instance.

awcullen, 2016-07-05:
Would you consider adding this to line 520?

`
function apply_filter(self, newDataValue) {
    
    if (!self.oldDataValue) {
        return true;
    }
    if (self.filter instanceof DataChangeFilter) {
        return apply_datachange_filter(self, newDataValue, self.oldDataValue);
    }
    // if filter not set, by default report changes to Status or Value only 
    return !sameDataValue(dataValue, self.oldDataValue, TimestampsToReturn.Neither);
}

`

erossignon, 2016-07-06:
may be here instead ? https://github.com/node-opcua/node-opcua/commit/0f923af34baf4158550c0af86309a244bceb665a#diff-6d82f75aa54392ae802d4a966b277aadR556

awcullen, 2016-07-06:
In my simulation server, my variables are recalculated at a set interval.  The nodes are bound to the variables using timestamped_get(). Some of the variable values do not change, but because the timestamp is updated, the server is sending them in every publish response.

I'd like to change that behavior so that it functions like the DataChangeTrigger.StatusValue if no filter is applied (or as you mentioned above, if no filter can be applied, in the case of 'String')

So I propose changing the final line to apply_filter()  

`
    // if filter not set, by default report changes to Status or Value only 
    return !sameDataValue(newDataValue, self.oldDataValue, TimestampsToReturn.Neither);

`
*** note that here I corrected the spelling of the 1st parameter, compared to my post above.

erossignon, 2016-07-14:
@awcullen , the behavior you described matches the specifications:

OPCUA 1.03 part 4 page 126 . 7.17.2 DataChangeTrigger 
> trigger Enum DataChangeTrigger
Specifies the conditions under which a data change notification should be reported. It has the following values:

|value|description|
|-----------|---------------|
|STATUS_0| Report a notification ONLY if the StatusCode associated with the value changes. See Table 173 for StatusCodes defined in this standard. Part 8 specifies additional StatusCodes that are valid in particular for device data.|
|STATUS_VALUE_1| Report a notification if either the StatusCode or the value change. The Deadband filter can be used in addition for filtering value changes. **This is the default setting if no filter is set.**|
|STATUS_VALUE_TIMESTAMP_2| Report a notification if either StatusCode, value or the SourceTimestamp change. If a Deadband filter is specified, this trigger has the same behaviour as STATUS_VALUE_1.|

>**If the DataChangeFilter is not applied to the monitored item, STATUS_VALUE_1 is the default reporting behaviour.**

closed 2016-08-14 by awcullen