node-opcua/node-opcua #222: TimeOut Error for connecting client to subscribe the data

arun4san, 2016-06-27:
Hi,
when we are try to connect the opc server and subscribe the parameter we are getting continuously timeout error in client,

ClientMonitoredItem#_monitor:  ERROR in createMonitoredItems   ServiceFault returned by server { /*ServiceFault*/
responseHeader                /* ResponseHeader                   */: {
  timeStamp                   /* UtcTime                          */: 2016-06-27T15:35:31.480Z
  requestHandle               /* IntegerId                        */: 3886               0xf2e
  serviceResult               /* StatusCode                       */: BadTooManyOperations (0x80100000)
  serviceDiagnostics          /* DiagnosticInfo                   */: {
    namespaceUri              /* Int32                            */: -1
    symbolicId                /* Int32                            */: -1
    locale                    /* Int32                            */: -1
    localizedText             /* Int32                            */: -1
    additionalInfo            /* String                           */: null
    innerStatusCode           /* StatusCode                       */: Good (0x00000)
    innerDiagnosticInfo       /* DiagnosticInfo                   */: {
    }
  }
  stringTable                 /* String                        [] */: [ ] (l=1)
  additionalHeader            /* ExtensionObject                  */: null
}

Please help me out to resolve this issue.

Version used : node-opcua": "0.0.54"

Thanks,
Arun S

erossignon, 2016-07-02:
In fact the server returns BadTooManyOperations.
What is the server ? can you share the code you use to connect to it ?

finalburner, 2017-03-01:
I have the same problem here, I'm trying to connect to a CODRA PANORAMA OPC UA Server.
Is there any mean to solve ?

erossignon, 2017-03-01:
please specify  node-opcua version

finalburner, 2017-03-02:
Actual version of node opc-ua is 3.10.10

finalburner, 2017-03-07:
Is there any solution or way of resolution ? 
Thanks in advance

erossignon, 2017-03-09:
node opc-ua 3.10.10  is not know here. Can you expose your server to the internet so we could conduct an investigation ?

finalburner, 2017-03-09:
In fact through the commande $ npm -v node-opcua, the version is 3.10.10.

The problem doen't occur when I subscribe to a limited number of variable.
But for my application I have a very large number ( ~20000 ) of variables to monitor and for now, I just opted to do then in one subscription. The error start to show just few moment after the program start to subscribe to variable of this big list.

Concerning the exposition of the server, I can do with but for a limited amount of time, So if u can just let me know when you have time so I can do It.

Thanks in advance.

erossignon, 2017-03-09:
```npm -v node-opcua`` is equivalent to ```npm -v``. It returns the version of ... ```npm```.

Try : ```npm ls node-opcua```instead.

The  ServiceFault is returned by the server. I suspect that the  problem comes form the Server. Hav e you submitted the issue to Codra ?

I would also suggest that you split your createMonitoredItem request into multiple calls...

finalburner, 2017-03-09:
Right, the version is 0.0.61.

I didn't yet, but as you say I should maybe split the monitored variables into multiple subscription.

If you have any other suggestion, that would be great.

erossignon, 2017-03-09:
Not necessarily multiple subscriptions. You can still have a single one, but gradually createMonitoredItem on this subscription a bunch at a time ( let say per batch of 1000 variables)

finalburner, 2017-03-09:
I'll try this, Thank you very much.

finalburner, 2017-03-14:
This is my actual source code, I've tried many times to dispatch the creation of monitored variables into mini-batches, but didn't succeed unfortunately. Hope you can do something. (Ids is a javascript object containing 25000 items)

ids.forEach(function(id){          

              adr = id.variablename ;
              adr +=  '_' + id.Information + '.Valeur';
               var nodeId = "ns=2;s=" + adr;
         var monitoredItem  = the_subscription.monitor({
           nodeId: opcua.resolveNodeId(nodeId),
           attributeId: opcua.AttributeIds.Value
       },
       {samplingInterval: 10, discardOldest: true, queueSize: 1 },
       opcua.read_service.TimestampsToReturn.Both            //didn't find docs of this 
       );

       monitoredItem.on("changed",function(dataValue){
    // do anything
       });
}

finalburner, 2017-04-19:
I've reached the OPC UA server developer who pointed to the paramater MaxQueuedRequestCount which was responsible for the error  BadTooManyOperations. 
I've set the parameter to a large value. BUT now it show another error :

erossignon, 2017-10-07:
* I would suggest to use ```async.eachLimit``` to iterate asynchronously the ids array , and prevent overwhelming the server with too many concurrent  indivudual monitored item creation.
* I would also suggest to use the '''subscription#monitorItems''' method
```javascript
               var itemsToMonitor = [
                    {
                        nodeId: resolveNodeId("ns=0;i=2258"),
                        attributeId: AttributeIds.Value
                    },

                    {
                        nodeId: resolveNodeId("ns=0;i=2258"),
                        attributeId: AttributeIds.Value
                    }
                ];
                var options = {
                    samplingInterval: 10,
                    discardOldest: true,
                    queueSize: 1
                };

                var monitoredItemGroup = subscription.monitorItems(itemsToMonitor, options);

                // subscription.on("item_added",function(monitoredItem){
                monitoredItemGroup.on("initialized", function () {
                    if (doDebug) {
                        console.log(" Initialized !");
                    }

                    monitoredItemGroup.monitoredItems.length.should.eql(2);

                    monitoredItemGroup.terminate(function () {
                        if (doDebug) {
                            console.log(" terminated !");
                        }
                        callback();
                    });
                });

```

closed 2017-10-07 by erossignon